carla-simulator/carla #3891: Carla's performance with multiple RGB sensors

amirtakouh, 2021-02-15:
Hello,

I am interested to know how to optimize Carla to run with an extensive set of sensors. I made some experiments with zero, two, six and ten RGB cameras and monitored the GPU and CPU utilizations. 
Carla version is 0.9.11 and it was running with the low quality level flag. 
A summary of the results:
zero cameras: 245 FPS, 260% CPU utilization and 50% GPU utilization.
two cameras: 45 FPS, 214% CPU utilization and 43% GPU utilization.
six cameras: 17 FPS, 170% CPU utilization and 38% GPU utilization.
ten cameras: 12 FPS, 145% CPU utilization and 38% GPU utilization.

So as the number of RGB cameras grows, the FPS falls (as expected), however both CPU and GPU utilization fall with it. My question is why isn't either CPU or GPU or both maxing out? I expected that either CPU or GPU becomes the bottleneck but seems like the bottle neck is somewhere else.
Could you please let me know your analysis of the mentioned data? Is this how you expect it to behave and if so, would you please elaborate?

corkyw10, 2021-03-15:
@DSantosO could you follow up on this please?

DSantosO, 2021-03-18:
Hello @amirtakouh,

We are now analyzing in detail the CPU usage of Carla so this data is very interesting. Are you using Linux or Windows? Sync or async mode? Are you monitoring also the communication? It could be that as more data is sent between client and server is sent it became the main bottleneck.

amirtakouh, 2021-03-18:
Hi @DSantosO,

Thank you for your response!
I am using Ubuntu 18.04 and running in async mode (default). The client and server are running on the same machine so I assume the communication (shared memory) is not a bottleneck.

Please let me know if more info is needed from my side.

raymanfx, 2021-03-29:
I'm also very interested in this.
We use a setup consisting of 3 cameras (front view 1280x720 and 2x side mirror views 720x720) and are getting ~10 FPS on a Ryzen 3 3700X + Nvidia RTX 2080 Ti.

I am able to allocate some of my team members for performance research, but would appreciate at least a hint on what are the main bottleneck here before wasting man hours finding them.

raymanfx, 2021-04-09:
@corkyw10 @DSantosO sorry for pinging you guys directly - is there any update here regarding some hints on performance bottlenecks? I have new team members starting on Monday and would be willing to spend some time on (RGB) sensor performance in Carla. Before doing so, I'd like to sync with you on known issues/bottlenecks so I can avoid wasting time profiling.

Axel1092, 2021-04-09:
Hi all, when using camera sensors, images are generated in the GPU and then they are copied to the CPU in order to be sent to the client. The GPU to CPU copy is an expensive operation which stalls both GPU and CPU, the larger the images or number of them, the longer the stall occurs producing the reduction in GPU and CPU utilization that you observe in your testing.
About profiling, we are improving client-server communication speed and we found some other performance issues in 0.9.11 that we are still investigating. 
Regardless, your bottleneck is just GPU -> CPU communication. A solution to improve performance would be to use multiple GPUs to render different cameras, this is still **not supported** by CARLA but it is in our roadmap.

raymanfx, 2021-04-09:
Would you mind sharing some more details on those "performance issues" you find in 0.9.11? Perhaps I can help with that investigation. As I said before - I'm willing to spend some man-hours here.

About the GPU to CPU copy - yeah we cannot avoid that. I still believe there are ways to optimize it especially for multiple sensors by e.g. batching copy operations for multiple cameras. I'm not sure how far down the stack this would need to be done though; I guess it's in one of the lower layers of the Unreal Engine.

I'm also interested in knowing how you are tackling the client-server communication speed issues. Do you plan to change your serialization pipeline or is it something else?

One last point: I'm quite interested in working with another language on the client side, not Python (nor C++). I'm looking at Rust in particular. Are there any plans to offer a C API for clients to talk to the server? I know the Python bindings are currently generated from C++ code using Boost, perhaps it is possible to start exposing some of those C++ calls as `extern C` with appropriate data structures? This is something for the long run for sure, just an idea I had in mind for quite a while now.

Axel1092, 2021-04-13:
About the performance issues @joel-mb can give you more details. 

About GPU and CPU copies, we are trying to find ways to improve it but as you say this is deep within the Unreal Engine code and a challenging problem on its own. 

The communication issues we found were related to how linux handles tcp information. By default there is an algorithm in place to manage when to send information optimally. This is nice for servers but in CARLA use case causes information packets to be delayed hurting performance in the process. We removed this delay and managed a noticeable improvement in medium-high fps situations (+30fps).

About the last point, this is not in our roadmap for now. Our focus is on the Python API and c++ interface.

closed 2023-06-02 by stale[bot]

zzj403, 2023-10-25:
> About the performance issues @joel-mb can give you more details.
>
> About GPU and CPU copies, we are trying to find ways to improve it but as you say this is deep within the Unreal Engine code and a [...]

I am very interested in the tcp delay removing feature, is there anyway to use this feature? thank you very much