carla-simulator/carla #3893: Segmentation fault

sulichuan, 2021-02-17:
HI
I installed Carla0.9.11 and Unreal Engine4.24
and successfully run Carla server with Town01 in UE4Editor
but when I tried to run "manual_control.py"
the script shows segmentation fault (core dumped)

the message is 
"WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4dc4cb81 
WARNING: Simulator API version  = 0.9.11 
Segmentation fault  (core dump)"

does anyone know how to solve it?

syveqc, 2021-02-17:
Seems like you are running and old version of the python API, maybe your pythonpath still points to an old API?

sulichuan, 2021-02-18:
@syveqc 
I had run"make PythonAPI" before running the Carla server, and it built successful.
Is there somewhere else I can change the python path?

syveqc, 2021-02-18:
I also have to set the PYTHONPATH environment variable every time I update the version, so you could try to do `echo $PYTHONPATH` and see whats in there, it should point to the .egg file in `PythonAPI/carla/dist`, mine is called `carla-0.9.11-py3.7-linux-x86_64.egg`, but that may vary if you built it yourself.

sulichuan, 2021-02-22:
@syveqc 
I followed the way what you suggest, and it works.
I can run vehicle in the server now.
Thanks for your helping.

syveqc, 2021-02-22:
Glad to hear that, no problem!

closed 2021-03-15 by corkyw10