carla-simulator/carla #3895: save_to_disk for PNG is slow

qhaas, 2021-02-17:
According to our benchmarks, at higher resolutions (e.g. 1080P), it is taking significantly longer to save as PNG when compared to JPEG and TIFF with CARLA 0.9.11.  While we expected TIFF to be the fastest by far since it is uncompressed, but we were expecting JPEG and PNG to perform similarly when it came to runtime.  From monitoring `top`, it would appear that a single CPU core is pegged to 100%, which might imply libpng isn't multi-threaded.  To reproduce, apply the attached patch to the 'sensor_synchronization.py' carla pythonapi example and run it.

```
World's frame: 5
    Frame: 5   Sensor: rgb
    tiff: 0.013288811001984868 seconds
    png: 0.9420106309989933 seconds
    jpeg: 0.025687530003779102 seconds

World's frame: 6
    Frame: 6   Sensor: rgb
    tiff: 0.01103343399881851 seconds
    png: 0.924331724003423 seconds
    jpeg: 0.025969877002353314 seconds

World's frame: 7
    Frame: 7   Sensor: rgb
    tiff: 0.012868407000496518 seconds
    png: 0.9671741249985644 seconds
    jpeg: 0.02668619000178296 seconds
```

This was conducted in an Ubuntu 18.04 environment with CARLA 0.9.11, a 4 physical core CPU, and a Pascal era quadro.

[format_times.txt](https://github.com/carla-simulator/carla/files/5995930/format_times.txt)

qhaas, 2021-02-19:
It would appear that this might be a result of libpng (by way of zlib) not supporting multi-threading.  The default action of [`boost::gil` is to compress PNGs](https://github.com/boostorg/gil/blob/boost-1.72.0/include/boost/gil/extension/io/png/tags.hpp#L203)

qhaas, 2021-08-25:
We ended up using background worker threads to perform the `save_to_disk` operation of select carla.sensordata classes, which allowed the simulation to progress while each cpu core was saving an image in the background.  Not perfect, but drastically improved our runtimes.

KevinLADLee, 2021-12-10:
Hi @qhaas , I also found this problem. I try to disable the compression when using png files, and it seems to work now.

https://github.com/KevinLADLee/carla/blob/221d244e6cfc832428e08d9cb1e57e0746723fb7/LibCarla/source/carla/image/ImageIOConfig.h#L122-L125

closed 2023-06-02 by stale[bot]

qhaas, 2025-01-20:
PNG seems to still be single threaded in CARLA 0.10.0, our performance increases by orders of magnitude when we are not saving out HD PNGs.

Sarang-Jokhio, 2025-01-24:
Hi @qhaas. I have CARLA 0.9.15 installed on my Linux system. I encountered issues saving PNG files, as Unreal Engine crashes when I try, even at lower resolutions. The simulation runs somewhat smoother with .jpg and .tiff files, but only at around 2K resolutions—it's not feasible at 4K, which is what I need. I have a quite decent PC configuration with NVIDIA GeForce RTX 4090.