carla-simulator/carla #3897: Strange behavior switching gear from R to N

robertprior-fixstars, 2021-02-18:
I'm seeing switching gear from R->N performs differently from other gear changes; rather than going R->N directly, gear instead goes R->1. Is this expected at least using manual transmission? I expected R->N change would function in the same way as N->1, N->R, 1->2 etc.

To reproduce:

1. start PythonAPI\examples\manual_control.py
2. press `m` turning on manual transmission
3. press `,` gear changes down from N->R
4. press `.` gear changes up from R->1 skipping N
5. press `w`,`a`,`s`, or `d` gear changes 1->N

This happens using Windows 10 / Carla 0.9.11 / Python 3.7 / pygame 2.0.0 
 
What is weird is that in this state changing any of brake, throttle or steer will set the gear to N (without using `,` or `.`). Any delta in these values during the gear change will cause gear to rapidly go R->1->N.

For example while car is stopped and in R hold `w`. Before the throttle reaches the maximum value press `.`. The gear changes rapidly R->1->N as the throttle keeps increasing.

Alternatively, let the throttle reach the maximum value and keep holding `w` and press `.`.  In this case, throttle is constantly at max value through the gear change. The car will switch R->1 and accelerate / move forward until `w` is released or `a`\\`s`\\`d` is pressed which causes the gear to switch 1->N and car will begin to decelerate.

The same behaviour occurs with steer / brake as well (if values are changing when `.` is pressed see R->1->N if steady see R->1).

corkyw10, 2021-03-15:
@Axel1092 could you follow up on this please?

robertprior-fixstars, 2021-05-20:
Hi any updates to this? I think it is a bug looking at it some more. Even continually using apply_control with gear set to 0 the gear will not change from 1 when in the above state without additional changes to brake etc.. I added simple debug prints to PythonAPI\examples\manual_control.py:496 like:

```
print("self ", self._control)
print("world", world.player.get_control())
world.player.apply_control(self._control)
```
to show the behaviour.

The control sequence looks like  (cropped out throttle, steer to avoid scrolling):

1. Both client and server match in R (gear is -1) - expected
```
self  VehicleControl(...brake=0.000000, hand_brake=False, reverse=True, manual_gear_shift=True, gear=-1)
world VehicleControl(...brake=0.000000, hand_brake=False, reverse=True, manual_gear_shift=True, gear=-1)
...
```
2. Client goes to 0 server has not updated yet - expected 
```
self  VehicleControl(...brake=0.000000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=0)
world VehicleControl(...brake=0.000000, hand_brake=False, reverse=True, manual_gear_shift=True, gear=-1)
...
```
3. Client stays at 0 but server changes to 1 - unexpected
```
self  VehicleControl(...brake=0.000000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=0)
world VehicleControl(...brake=0.000000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=1)
...
```
4.  Brake changes causes server gear to update- unexpected
```
self  VehicleControl(...brake=0.400000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=0)
world VehicleControl(...brake=0.000000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=1)
...
world VehicleControl(...brake=0.400000, hand_brake=False, reverse=False, manual_gear_shift=True, gear=0)
```

robertprior-fixstars, 2021-06-25:
Sorry to keep the thread going with only my input here but the root cause of above is this line:

https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp#L240

```
if (LastAppliedControl.bReverse != InputControl.Control.bReverse)
    {
      MovementComponent->SetUseAutoGears(!InputControl.Control.bReverse);
      MovementComponent->SetTargetGear(InputControl.Control.bReverse ? -1 : 1, true); <-------
```

I am not sure if there is a reason for doing the -1 -> 1 gear change. In my use case, I always want the gear R->N change to work so swapping 1 to 0 in that line fixes the problem for me. Presumably however, this would break any R-> non N  gear change in the same way as the original code behaved for non R->1 changes. So I am not sure what the proper fix for this would be.

closed 2023-06-02 by stale[bot]