carla-simulator/carla #3900: [Testing]

XGodina, 2021-02-18:
That you have to test

- Check how to see the distance of the camera.

1. - Motorbikes
2. - Bicycles
3. - Pedestrian
4. - Cars
5. - Van
6. - Assets

XGodina, 2021-02-18:
I've tested the vehicles and I've seen one of the cameras on bicycles didn't see well.

XGodina, 2021-02-18:
[image]
[image]

XGodina, 2021-02-18:
And I think the camera on the bike could be better.

DSantosO, 2021-02-18:
You are right, I have modified the lateral camera to let the x parameter fixed, and then it is closer to what it was before.

[image]

XGodina, 2021-03-03:
[image]
[image]
[image]

I've tested de camera and I think the camera of pedestrians is far away from it. (Picture 1 and Picture 2)
Picture 3 It would be better if the camera position was 90 degrees overt to the pedestrian. This way we could better see the animations in silhouette. The current position doesn't help us.

closed 2021-09-03 by DSantosO