carla-simulator/carla #3902: Can Not Add Third Car

arnabfly, 2021-02-18:
Below, I add my code. I can add two vehicles (vehicle and vehicleE). But when I am trying to add thrid one, its automatically destroying (vehicleA). Can anyone tell me how to solve this?
    try:
        m = world.get_map()
        start_pose = random.choice(m.get_spawn_points())
        waypoint = m.get_waypoint(start_pose.location)

        blueprint_library = world.get_blueprint_library()

        try:

            vehicle = world.spawn_actor(
                random.choice(blueprint_library.filter('vehicle.*')),
                start_pose)
            actor_list.append(vehicle)
            vehicle.set_autopilot(True)
            print("1st done")
        except:
            print("1st not done")

        try:
            vehicleE = world.spawn_actor(
                random.choice(blueprint_library.filter('vehicle.*')),
                start_pose)
            print("2nd Done")
        except:
            print("2nd not done")

        try:
            vehicleA = world.spawn_actor(
                random.choice(blueprint_library.filter('vehicle.*')),
                start_pose)
            print("3rd Done")
        except:
            print("3rd not done")

        camera_rgb = world.spawn_actor(
            blueprint_library.find('sensor.camera.rgb'),
            carla.Transform(carla.Location(x=-5.5, z=2.8), carla.Rotation(pitch=-15)),
            attach_to=vehicle)
        actor_list.append(camera_rgb)

syveqc, 2021-02-22:
Are you sure this works? You are trying to spawn three vehicles on exactly the same location, so you should get a "Spawn Failed" error! Your script works for me when I am using deterministic, non-colliding spawn positions. 
Also a general advice: You get a lot more feedback why your code does not work if you print/log the error messages instead of just catching them and doing nothing with them!

corkyw10, 2021-03-15:
@arnabfly has your issue been resolved?

closed 2021-08-05 by corkyw10