carla-simulator/carla #3903: carla-0.9.11 reported errors during operation

whuxhzhang, 2021-02-19:
ubuntu18.04
carla-0.9.11

I compiled the source code of carla-0.9.11, and run CarlaUE4.sh, but find these errors: 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=122688 LargeMemoryPoolOffset=253824 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

qhaas, 2021-02-19:
The community will likely need a tad more to go on...

1. Which GPU are you using?  If nvidia, does `nvidia-smi` produce an error?  This will determine if you are capable of 3D acceleration with your GPU
2. Are you running headless and/or X-less (i.e. no x-server)?  If so, you should use [opengl](https://carla.readthedocs.io/en/latest/adv_rendering_options/#setting-off-screen-mode), and start carla like `SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl`

whuxhzhang, 2021-02-20:
>     1. nvidia-smi

[image]

All town_maps are available, including my own. Only one map I made had this problem. So I think it's not the GPU's fault.

corkyw10, 2021-03-15:
@bernatx could you have a look at this please?

yz-bry, 2021-10-06:
I have the same problem as you.

qhaas, 2021-12-09:
I've started seeing this in CARLA 0.9.12, but only with jobs launched via SLURM, and with the 'default' maps even, works fine on 'bare metal' (i.e. just ssh into the box and run it):
```
./CarlaUE4.sh -nosound -vulkan -RenderOffscreen -graphicsadapter=0 -carla-rpc-port=2050
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
```

xbowlove, 2023-01-13:
@yz-bry have you solved the problem? I face this problem these days.if solved, could you please tell me how to deal with it.

paulvantieghem, 2023-04-08:
> The community will likely need a tad more to go on...
>
> 1. Which GPU are you using?  If nvidia, does `nvidia-smi` produce an error?  This will determine if you are capable of 3D acceleration with [...]

Running `SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -opengl` instead of `./CarlaUE4.sh -opengl -nosound` worked for me on 0.9.11