carla-simulator/carla #3904: Carla is not starting on Windows 10 - required DirectX Runtime (which is preinstalled)

cloud-rocket, 2021-02-19:
The following error reported: 

"The following component(s) are required to run this program: DirectX Runtime"

Tried with precompiled: CARLA_0.9.11 and CARLA_0.9.9.4.zip

DxDiag.exe details:
* Operating System: Windows 10 Pro Insider Preview 64-bit (10.0, Build 21301) (21301.rs_prerelease.210123-1645)
* DirectX Version: DirectX 12
* Card name: NVIDIA GeForce RTX 3060 Ti (latest preview driver)

cloud-rocket, 2021-02-19:
Ok, I found an old installation of DirectX 11 here - https://www.filehorse.com/download-directx-11/download/
And it works

closed 2021-02-19 by cloud-rocket

munila, 2022-02-07:
Hi cloud-rocket,
i am facing the same issue with the same version of directX and carla. Where exactly did you put the old directX installation to make carla run?
many thanks in advance

munila, 2022-02-08:
if someone faces the same issue the following approach was the solution for me:

[devinyang1992](https://forum.carla.org/u/devinyang1992)

[May '20](https://forum.carla.org/t/directx-runtime-error/214/2)

It is really weird. I encountered this error before, it worked when i did it like this way:
Download 0.9.4 version, run CarlaUE4.exe, this time it would say you do not have DirectX Runtime, then you got a option to click to fix it automatically.
hope this will help you.

Kerstin-Keller, 2022-06-24:
This was the case for me, too.
Can this please be adressed it the installation guide, and also the installers for the prerequisites be made available either with the  installer or to be downloaded separately?
This is not  a good "getting started" experience 😞

sanadaryo, 2023-07-24:
Hello everyone,

I am also encountering the same issue as described in this thread. I am trying to run Carla on a Windows Server 2019 instance on AWS EC2 g4dn.xlarge, which comes with an NVIDIA Tesla T4 GPU. However, when I try to run CarlaUE4.exe, I encounter an error message stating "DX11 feature level 10.0 is required to run the engine".

My system supports DirectX 12, and when I run dxdiag, it shows DirectX 12. Also, I have updated my NVIDIA drivers to the latest version.

I have tried the solutions suggested in this thread, but they did not resolve the problem. Specifically, I downloaded and installed an older DirectX 11 installer and tried Carla version 0.9.4, but I still get the same error message.

Additionally, I am using NICE DCV for remote desktop connection, which should properly expose the GPU.

If anyone has any advice on this issue, please let me know. Any suggestions are welcome.

Thank you in advance.