carla-simulator/carla #3906: Braking Behaviour for very low speed (< 1m/s)

i14d4, 2021-02-19:
Hello CARLA team,

I am wondering if I could improve / fix the following described behaviour through e.g. specific parameters within the UE settings or if you could give me a brief explanation why it is occurring.

I noted that the velocity of a vehicle drops very fast at some point of time when the vehicle reachs a very low speed. This behaviour occurs with different vehicles and also with very small brake command values like 0.001.
By changing the setting of the fixed_delta_seconds it drastically changes at which point of velocity this "velocity drop" occurs. So I assume that it is related to the simulation physics calculation at very low speed.
Is there any chance to improve this behaviour through tuning settings within UE to have a better velocity simulation for low speed (apart from setting the simulation delta time to a very low value) ?
 
To plot the above described I used the script of @hh0rva1h (https://github.com/carla-simulator/carla/issues/3256) and changed the settings as follows:

Setting: fixed_delta_seconds = 0.045, Vehicle: Tesla Model 3, Brake command: throttle: 0.0, brake: 0.0
[image]
[image]

Setting: fixed_delta_seconds = 0.045, Vehicle: Tesla Model 3, Brake command: throttle: 0.0, brake: 0.001
[image]
[image]

Setting: fixed_delta_seconds = 0.00045, Vehicle: Tesla Model 3, Brake command: throttle: 0.0, brake: 0.001
[image]

Thanks in advance for any help !

corkyw10, 2021-03-15:
@jackbart94 could you follow up on this please?

jackbart94, 2021-03-17:
> I noted that the velocity of a vehicle drops very fast at some point of time when the vehicle reachs a very low speed. This behaviour occurs with different vehicles and also with very small brake command values like 0.001.
> By changing the setting of the fixed_delta_seconds it drastically changes at which point of velocity this "velocity drop" occurs. So I assume that it is related to the simulation physics calculation at very low speed.
> Setting: fixed_delta_seconds = 0.00045, Vehicle: Tesla Model 3, Brake command: throttle: 0.0, brake: 0.001

If you change so drastically the `fixed_delta_seconds` setting, physics won't work properly. They are calibrated to work at 20 FPS, so please keep that value to about 0.05.

i14d4, 2021-03-17:
Hello @jackbart94 

the same behaviour is observed with values of 0.05 for fixed_delta_seconds.
The first screenshots in my first post are with fixed_delta_seconds of ~0.05. Only for further checking I used lower delta_seconds.

(e.g. fixed_delta_seconds = 0.045, Vehicle: Tesla Model 3, Brake command: throttle: 0.0, brake: 0.001)

thanks in advance for any help or explanation!

closed 2021-11-18 by jackbart94

DrShIkIgAmy, 2023-04-04:
Hi there) I understand that carla is opensource and it's provided AS IS, but I think it is not right to stale each issue you cannot resolve. I have the same issue, and as it's mentioned before, I use fixed_delta_seconds == 0.05. So could you clue me what to do?

oliver-schumann, 2024-11-07:
Any updates on this issue? It's getting quite frustrating that this is still not fixed in 0.9.15. 

I hope the rework to 0.10.0 will fix it but it is a pitty, that a bug like this is still around as it makes the simulation of slow parking maneuvers impossible.

Everybody I talk to on conferences etc. had this issue and had to replace the blueprint simulating the vehicle by themselves or even use an external vehicle model and respawn the vehicle with deactivated physics.
Unfortunately, all these approaches are very tedious and time consuming...

AKG47-Auto, 2025-04-10:
> Any updates on this issue? It's getting quite frustrating that this is still not fixed in 0.9.15.
>
> I hope the rework to 0.10.0 will fix it but it is a pitty, that a bug like this is still around as [...]

Can you share some more details on the custom blueprint approach

oliver-schumann, 2025-04-10:
Unfortunately no.
I just avoid this issue by suppressing break maneuvers under 1 m/s. The friction then deaccelerates the vehicle further. When I reach the goal, I allow breaking again which leads to an instant stillstand at my goal.
This is super dirty but the best I could do for now. 

You could also increase the friction further to allow the vehicle to always be in a scenario where it either coasts or presses throttle without needing to press the brake in he low speed cases.

Changing the blueprint properties containing and the used physics parameters did not help  unfortunately.

AKG47-Auto, 2025-04-11:
> Unfortunately no. I just avoid this issue by suppressing break maneuvers under 1 m/s. The friction then deaccelerates the vehicle further. When I reach the goal, I allow breaking again which leads to an instant stillstand at my goal. This is super dirty but the best I could do for now.
> 
> You could also increase the friction further to allow the vehicle to always be in a scenario where it either coasts or presses throttle without needing to press the brake in the low speed cases.
> 
> Changing the blueprint properties containing and the used physics parameters did not help unfortunately.

yeah editing the blueprint didn't help me as well. I am trying to train an agent to drive while maintaining a certain threshold of body slip so I need the brakes to work. I guess I'll try with discrete action space and see how it goes