carla-simulator/carla #3910: ImportError: cannot import name 'ad' from 'carla'

saina110, 2021-02-21:


corkyw10, 2021-03-15:
Hi @saina110, have you installed the dependencies and build the components according to this [documentation](https://carla.readthedocs.io/en/latest/adv_rss/#dependencies)?

ga32riv, 2021-07-13:
"~/carla/PythonAPI/carla/dist$ sudo python3.7 -m easy_install carla-0.9.11-py3.7-linux-x86_64.egg"       solved it for me.

closed 2021-08-05 by corkyw10

chenhengwei1999, 2022-05-03:
> **Ubuntu: 18.04 LTS Kernel: 5.4.0-42-generic Graphic Card: Nvidia GTX 1650ti Carla simulator: 0.9.11 Python:3.7.9**
> 
> Dear everyone,
> 
> I am trying to run the file "manual_control_rss.py" under "PythonAPI/examples/rss" folder. When I typed "python3 manual_control_rss.py", the terminal shows me the error: `ImportError: cannot import name 'ad' from 'carla' (/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.11-py3.7-linux-x86_64.egg/carla/__init__.py)`
> 
> I have already downloaded the "ad-rss-lib" into the whole "Carla_simulator" folder. But I didn't do any other thing.
> 
> Very appreciated anyone could provide me with some hints or suggestions! Thanks in advance!
> 
> Best Regards

Hi, guys, Hello, friend, have you solved it now?