carla-simulator/carla #3914: Carla ERROR: Cook failed. (when make package)

saina110, 2021-02-22:
**Ubuntu: 18.04 LTS
Kernel: 5.4.0-42-generic
Graphic Card: Nvidia GTX 1650ti
Carla simulator: 0.9.11
Python:3.7.9**

Dear everyone,

I am trying to run "make package" in order to get executable CarlaUE4 file, but I met with the "cook fail" termination. In the log file, it says: 

Log.WriteException: **ERROR: Cook failed.**
Log.WriteException:        (see /home/yuan/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
Log.WriteException: 
**Log.WriteException: CommandletException: Editor terminated with exit code 137 (signal 9)** while running Cook for /home/yuan/Project/carla/Unreal/CarlaUE4/CarlaUE4.uproject; see log /home/yuan/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2021.02.22-13.44.51.txt

I searched some answers from the website, some said that '137' code means out of memory.. Is that true? How could I fixed this problem ? I am very confused about this issue

Thank you for your suggestions and look forward to your feedback

Very appreciated

Best Regards

corkyw10, 2021-03-15:
@bernatx could you have a look at this please?

NielsPichon, 2021-06-03:
@saina110 can you see something like `Display: Max memory allowance for cook 16384mb min free memory 0mb` in your logs? If so, you are right in thinking you are out of memory. Else you can ctrl-f in the logs for `error` or `failed` which may indicate some possible errors.

closed 2023-06-02 by stale[bot]