carla-simulator/carla #3915: 'make LibCarla.client.rss' failed

saina110, 2021-02-22:
Dear everyone, 

I am trying to build RSS package, but when I use the command: "make LibCarla.client.rss", it shows me the following error:

Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.11-4-gfe3cb686-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
Ad-rss.sh: ad-rss-4.4.0 for python3 already installed.
BuildLibCarla.sh: Building LibCarla "ClientRSS.Debug" configuration.
ninja: **error: '../ad-rss-4.4.0/install/ad_physics/lib/libad_physics.a', needed by 'LibCarla/cmake/test/libcarla_test_client_debug', missing and no known rule to make it**
Util/BuildTools/Linux.mk:128: recipe for target 'LibCarla.client.rss.debug' failed
make: *** [LibCarla.client.rss.debug] Error 1

There is no much useful information on the website to solve this problem. I am very appreciated that you could provide me with some hints and suggestions.

Thanks so much in advance!

Best Regards

saina110, 2021-02-24:
anyone could give me some suggestions? thanks so much! I am stuck at this issue for few days...

jackbart94, 2021-02-25:
Sorry for keeping you waiting. We are having a similar problem, so we are looking into fixing this!
We will let you know soon, thanks for finding this out!

jackbart94, 2021-02-25:
It's been failing for Python 3.6, but working fine with Python 3.7 and 3.8. Which version are you using?

saina110, 2021-02-26:
> It's been failing for Python 3.6, but working fine with Python 3.7 and 3.8. Which version are you using?

Hi, thank you for your support. My current version is:

**Ubuntu: 18.04 LTS
Kernel: 5.4.0-42-generic
Graphic Card: Nvidia GTX 1650ti
Carla simulator: 0.9.11
Python:3.7.9**

I have successfully built rss_lib by command "colcon build", when i jump to the command "make libcarla.client.rss" it still shows me the above error

jackbart94, 2021-02-26:
Could you try with Python 3.8 as well? I had your same error and following the documentation with Python 3.8 I was able to make it work.

saina110, 2021-02-26:
> Could you try with Python 3.8 as well? I had your same error and following the documentation with Python 3.8 I was able to make it work.

I update the python3.7 to python3.8, and reboot the system. then when I directly type 'make LibCarla.client.rss', the console shows the same error like above.. Do I need to make additional configurations?

`BuildLibCarla.sh: Building LibCarla "ClientRSS.Debug" configuration.
ninja: error: '../ad-rss-4.4.0/install/ad_physics/lib/libad_physics.a', needed by 'LibCarla/cmake/test/libcarla_test_client_debug', missing and no known rule to make it
Util/BuildTools/Linux.mk:128: recipe for target 'LibCarla.client.rss.debug' failed
make: *** [LibCarla.client.rss.debug] Error 1
`
current python version is: **Python3.8.0**

Thanks in advance

jackbart94, 2021-03-02:
Follow carefully what it says at [this link](https://carla.readthedocs.io/en/latest/adv_rss/) please.

saina110, 2021-03-03:
> Follow carefully what it says at [this link](https://carla.readthedocs.io/en/latest/adv_rss/) please.

I redo the steps following the link you share:
**Colon Build**
(1). **colcon build** within the **ad-rss-lib** folder:
`Starting >>> spdlog  
Starting >>> ad_map_opendrive_reader
Starting >>> PROJ4
Finished <<< spdlog [0.15s]                                                                                                        
Starting >>> ad_physics
Finished <<< ad_map_opendrive_reader [0.19s]                                                                                   
Finished <<< PROJ4 [0.21s]
Finished <<< ad_physics [0.18s]                   
Starting >>> ad_map_access
Starting >>> ad_rss
Finished <<< ad_rss [0.20s]                                                                   
Finished <<< ad_map_access [0.31s]                   
Starting >>> ad_map_access_qgis_python
Starting >>> MapMaker
Starting >>> ad_map_access_test_support
Starting >>> ad_rss_map_integration
Finished <<< ad_map_access_test_support [0.11s]                                                            
Finished <<< MapMaker [0.17s]                                                                                
Finished <<< ad_rss_map_integration [0.25s]                                                                            
Finished <<< ad_map_access_qgis_python [3.55s]                      
Starting >>> ad_map_access_qgis
Finished <<< ad_map_access_qgis [0.06s]                  

Summary: 11 packages finished [4.67s]
`
(2): command **colcon build --cmake-args -DBUILD_TESTING=ON**

`Summary: 11 packages finished [1.49s]
  3 packages had stderr output: PROJ4 ad_map_access_qgis ad_map_opendrive_reader
`
(3) **Colcon test**:

`Starting >>> spdlog  
Starting >>> ad_map_opendrive_reader
Starting >>> PROJ4
Finished <<< ad_map_opendrive_reader [0.04s]
Finished <<< spdlog [0.06s]
Starting >>> ad_physics
Finished <<< ad_physics [0.20s]                                      
Starting >>> ad_map_access
Starting >>> ad_rss
Finished <<< PROJ4 [0.42s]                                                              
[Processing: ad_map_access, ad_rss]                                         
Finished <<< ad_map_access [47.2s]                                          
Starting >>> ad_map_access_qgis_python
Starting >>> MapMaker
Starting >>> ad_map_access_test_support
Finished <<< ad_map_access_qgis_python [0.00s]
Starting >>> ad_map_access_qgis
Finished <<< MapMaker [0.00s]
Finished <<< ad_map_access_test_support [0.00s]
Finished <<< ad_map_access_qgis [0.00s]
Finished <<< ad_rss [1min 2s]                   
Starting >>> ad_rss_map_integration
Finished <<< ad_rss_map_integration [9.11s]                  

Summary: 11 packages finished [1min 12s]`

(4) **colcon build --metas colcon_python.meta**:  
`Starting >>> spdlog  
Starting >>> ad_map_opendrive_reader
Starting >>> PROJ4
Finished <<< spdlog [0.66s]                                                                                                      
Starting >>> ad_physics
--- stderr: ad_physics                                                                                                           
[... 42 lines cut ...]
Then, the next step: **make LibCarla.client.rss**:

`Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.11-4-gfe3cb686-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
Ad-rss.sh: ad-rss-4.4.0 for python3 already installed.
BuildLibCarla.sh: Building LibCarla "ClientRSS.Debug" configuration.
ninja: error: '../ad-rss-4.4.0/install/ad_physics/lib/libad_physics.a', needed by 'LibCarla/cmake/test/libcarla_test_client_debug', missing and no known rule to make it
Util/BuildTools/Linux.mk:128: recipe for target 'LibCarla.client.rss.debug' failed
make: *** [LibCarla.client.rss.debug] Error 1
`
It shows me the same error I put in the post...

do you think the problem is coming from the python binding?

Thank you for your time to support

Best Regards

saina110, 2021-03-11:
> Follow carefully what it says at [this link](https://carla.readthedocs.io/en/latest/adv_rss/) please.

Hi, do you have some suggestions on my previous test? did you also follow these steps without any errors? thank you so much ! 

Best Regards,

jackbart94, 2021-03-17:
> > Follow carefully what it says at [this link](https://carla.readthedocs.io/en/latest/adv_rss/) please.
> 
> Hi, do you have some suggestions on my previous test? did you also follow these steps without any errors? thank you so much !
> 
> Best Regards,

As I've said, with Python 3.8 I've had no problem whatsoever. In your message I haven't seen anything that suggests you are using 3.8, so please create a virtual environment with Python 3.8 and start from scratch.

closed 2021-11-18 by jackbart94

Daraan, 2024-02-20:
This Issue was matching my error. Despite using Python3.8, or any other in the range of 3.7-3.10, the error persisted.
For the future reader: My solution was to use the dev branch above 0.9.15 which has the necessary fixes for the RSS build. I think a later RSS version with internal fixes is necessary.

---

Some more (but incomplete) details on the problem:

I could track it down to the build of **ad-rss-4.4.3 failing to build at the first execution of `LibCarla.client.rss`.**
Trying it a ****second time** states: `Success! Ad-rss.sh: ad-rss-4.4.X for python3 already installed.` followed by the error reported above.
**The detection that ad-rss has been built successfully is wrong. Leading to the error that the named file cannot be found.**

---

Within ad-rss I realized that it is related to the dependencies and submodules. I did not write it down. Going to a later ad-rss version enabled me to build ad-rss sucessfully with Python as a standalone.

Daraan, 2024-04-12:
Dear Harsh & others,

To be more precise I was able to build Carla with RSS based on this commit 1433186a01963c1e6b66621427d9e8bfbf3c8208 hope this works for you too.
I do not think relevantly changed is my branch https://github.com/Daraan/CARLA-Simulator-Fork-DebianFix/tree/MyDevVersion where I was able to compile everything.

Secondly I composed a branch https://github.com/Daraan/CARLA-Simulator-Fork-DebianFix/tree/rss-python3.10-working 
based on 0ac122b69921c3d1454e0f2eb7ad0c83a5283f44 with an additional commit 7003a7cfbf3045b305a163f7e9a877a0850b7a5a (Ad-rss-4.5.3)
**but omitting 800b394d2ade9f76c191832484470ed26571e8be ** (downgrading to 4.4.4). However, I do not remember the reason why I did it.

I just tested this branch for me it compiles correctly using Python3.10 without and with RSS (both PythonAPI and client).

---

**Important:** If you run into an error when installing `ad-rss` be sure to delete the `ad-rss-4.X.X` folder and try again, else you might face different errors.
If you face an error during compiling ad-rss try to solve it, else you cannot continue.

wardstoneX, 2024-11-23:
> Dear Harsh & others,
> 
> There are two branches I was able to compile RSS sucessfully:
> 
> 1. based on this commit [1433186](https://github.com/carla-simulator/carla/commit/1433186a01963c1e6b66621427d9e8bfbf3c8208) hope this works for you too.
>    1.5) I do not think relevantly changed is my branch https://github.com/Daraan/CARLA-Simulator-Fork-DebianFix/tree/MyDevVersion where I was able to compile everything. I am using this branch with Python 3.7 & mainly 3.10.
> 2. I composed a branch https://github.com/Daraan/CARLA-Simulator-Fork-DebianFix/tree/rss-python3.10-working
>    based on [0ac122b](https://github.com/carla-simulator/carla/commit/0ac122b69921c3d1454e0f2eb7ad0c83a5283f44) with an additional commit [7003a7c](https://github.com/carla-simulator/carla/commit/7003a7cfbf3045b305a163f7e9a877a0850b7a5a) (Ad-rss-4.5.3)
>    **but omitting [800b394](https://github.com/carla-simulator/carla/commit/800b394d2ade9f76c191832484470ed26571e8be)** (downgrading to 4.4.4). However, I do not remember the reason why I did it.
> 
> I just tested this branch for me it compiles correctly using Python3.10 without and with RSS (both PythonAPI and client).
> 
> **Important:** If you run into an error when installing `ad-rss` be sure to delete the `ad-rss-4.X.X` folder and try again, else you might face different errors. If you face an error during compiling ad-rss try to solve it, else you cannot continue.
> 
> I am not sure how relevant it is but my build order normally is this, trying PythonAPI.rss before PythonAPI sometimes did not work.
> 
> ```
> # Test the basics
> make launch
> make LibCarla
> make PythonAPI
> #
> make LibCarla.client.rss
> # Sometimes I do make PythonAPI here
> make PythonAPI.rss
> ```

I was having the same problem, his 1st solution worked for me with python 3.8 and i had to include a linking flag -ldl to the build command to get it working...