carla-simulator/carla #3916: Added a vehicle, but there was a physical error.

peacekim86, 2021-02-23:
Hi, there.

I added a new vehicle in the picture. ( 4-wheels )
[image]

But it has a physics error that it tilts toward the front wheels(Like handstand ) when running the CARLA. (or running with python clinet)
[image]

Can you tell me why this is happen?

Additional, I made a vehicle at Blender. (+rigging)

jrefi, 2021-09-01:
Any update on this issue? I am seeing the same thing with my vehicle. I also used Blender to module the vehicle and this [add-on](https://continuebreak.com/creations/ue4-vehicle-rigging-addon-blender/) to add the rigging.

[image]

jrefi, 2021-09-07:
Figured it out. The Center of Mass of my vehicle was somehow offset to be in front and below the mesh. After fixing the Center of Mass Offset