carla-simulator/carla #3917: Scenic requires Python > 3.8 causes conflict with CARLA's dependence on Python 3.7

AntiLibrary5, 2021-02-23:
CARLA version == 0.9.11
Python version == 3.8.8

Hi, I have been trying to use the scenic API for CARLA. But having followed the instructions at: 
https://scenic-lang.readthedocs.io/en/latest/simulators.html#simulators
I run into error which of python version conflict.
Previously I have tried using Python 3.8 for CARLA but it didn't work, but worked only with v3.7. But now scenic's hard dependence on v3.8 is problematic. Is there a workaround to this problem?

I have been using CARLA with a conda venv with Python 3.7 and it works fine.
For scenic, I created a new venev with Python 3.8 and 
>> pip install scenic 
>> easy_install /PATH_TO_CARLA_FOLDER/PythonAPI/carla/dist/carla-0.9.11-py3.7-win-amd64.egg

Upon running: >> scenic [file-name]
I get the following error:

import carla as _carla
File "c:\users\vaibh\anaconda3\envs\venv_scenic\lib\site-packages\carla-0.9.11-py3.7-win-amd64.egg\carla\__init__.py", line 8, in <module>
    from .libcarla import *
  File "c:\users\vaibh\anaconda3\envs\venv_scenic\lib\site-packages\carla-0.9.11-py3.7-win-amd64.egg\carla\libcarla.py", line 7, in <module>
    __bootstrap__()
  File "c:\users\vaibh\anaconda3\envs\venv_scenic\lib\site-packages\carla-0.9.11-py3.7-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
    imp.load_dynamic(__name__,__file__)
  File "c:\users\vaibh\anaconda3\envs\venv_scenic\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: Module use of python37.dll conflicts with this version of Python.

glopezdiest, 2021-02-26:
Hey @AntiLibrary5, here is an egg file for windows and Python 3.8 ([carla-0.9.11-py3.8-win-amd64](https://github.com/carla-simulator/carla/files/6049069/carla-0.9.11-py3.8-win-amd64.zip)), see if your problems are solved. This can be easily created if you are using CARLA from source but not from packages.

Anyway, that indeed raises a problem, so we'll see if, from now on, it is viable to also put python3.8 eggs at our next releases.

AntiLibrary5, 2021-02-26:
Hi,
Thank you. With the egg file for python3.8, I was able to run scenic.

Indeed having the egg file for python3.8 in the packaged installation would be great since (at least for me due to storage constraints) building CARLA from source is a bit troublesome.

For anyone with similar issue, after downloading the egg file, in your venv:
> easy_install [CARLA-path]/PythonAPI/carla/dist/carla-0.9.11-py3.8-win-amd64.egg

closed 2021-02-26 by AntiLibrary5

harsha-vk, 2021-07-09:
> Hey @AntiLibrary5, here is an egg file for windows and Python 3.8 ([carla-0.9.11-py3.8-win-amd64](https://github.com/carla-simulator/carla/files/6049069/carla-0.9.11-py3.8-win-amd64.zip)), see if [...]

Can you provide py3.8 egg file for ubuntu 20.04

yhhSTM05, 2022-02-14:
> Hey @AntiLibrary5, here is an egg file for windows and Python 3.8 ([carla-0.9.11-py3.8-win-amd64](https://github.com/carla-simulator/carla/files/6049069/carla-0.9.11-py3.8-win-amd64.zip)), see if [...]

Hi @glopezdiest, sorry for disturbing you,
I met the same problem with @AntiLibrary5.  I also ran into error which of python version conflict (CARLA version == 0.9.8
Python version == 3.8.5). If possible, can you give me some guidence on generating the 'egg file' using CARLA from source, or help generate it directly, thanks very much for your helps! Many thanks!

yhhSTM05, 2022-02-14:
> 

Many thanks for your advice, but this method is not work. Maybe the right way is generate the egg file from the source, but I an not sure how to create it through the Carla from source.

yhhSTM05, 2022-02-14:
> > > Hey @AntiLibrary5, here is an egg file for windows and Python 3.8 ([carla-0.9.11-py3.8-win-amd64](https://github.com/carla-simulator/carla/files/6049069/carla-0.9.11-py3.8-win-amd64.zip)), see if your problems are solved. This can be easily created if you are using CARLA from source but not from packages.
> > > Anyway, that indeed raises a problem, so we'll see if, from now on, it is viable to also put python3.8 eggs at our next releases.
> > 
> > 
> > Hi @glopezdiest, sorry for disturbing you, I met the same problem with @AntiLibrary5. I also ran into error which of python version conflict (CARLA version == 0.9.8 Python version == 3.8.5). If possible, can you give me some guidence on generating the 'egg file' using CARLA from source, or help generate it directly, thanks very much for your helps! Many thanks!
> 
> Try renaming the .egg file with your python version.

Many thanks for your advice, but this method is not work. Maybe the right way is generate the egg file from the source, but I an not sure how to create it through the Carla from source.

jasmeetgaba24, 2022-08-19:
> [0.9.11-py3.8-win-amd64](https://github.com/carla-simulator/carla/files/6049069/carla-0.9.11-py3.8-win-amd64.zip)), see if your problems are solved. This can be easily created if you are using CARLA from source but

+1