carla-simulator/carla #3918: REQUIRED process [carla_ros_bridge-1] has died!

sulichuan, 2021-02-23:
Hi~
I had  installed the ROS bridge, but when I trying to run it with "roslaunch carla_ros_bridge carla_ros_bridge.launch"
The script showed "REQUIRED process [carla_ros_bridge-1] has died!", and shut down 

the message is
================================================================================REQUIRED process [carla_ros_bridge-1] has died!
process has died [pid 26150, exit code 1, cmd /home/c/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py __name:=carla_ros_bridge __log:=/home/c/.ros/log/84c9eacc-75c4-11eb-b62a-7085c2c5085a/carla_ros_bridge-1.log].
log file: /home/c/.ros/log/84c9eacc-75c4-11eb-b62a-7085c2c5085a/carla_ros_bridge-1*.log
Initiating shutdown!
================================================================================

the carla version is 0.9.11 
ROS bridge version not sure, I just followed the step "b) Using source repository", and set the pannel in the bottom right corner of the window into 0.9.9

does anyone knows how to solve it?

corkyw10, 2021-03-15:
@joel-mb could you have a look at this please?

joel-mb, 2021-03-16:
Hi @sulichuan!
Right now the CARLA ROS bridge works only with CARLA `0.9.10` and `0.9.10.1`. However, we've been testing it with CARLA `0.9.11` and everything seems to work well. If you want to use it with `0.9.11` just change the following line in `bridge.py` (https://github.com/carla-simulator/ros-bridge/blob/master/carla_ros_bridge/src/carla_ros_bridge/bridge.py#L54) to the proper version.

Feel free to reopen the issue if the error persists.

closed 2021-03-16 by joel-mb