carla-simulator/carla #3920: dev.i915.perf_stream_paranoid=0

maria8899, 2021-02-23:
Hi,

I am trying CARLA 0.9.11 and CARLA 0.9.10.1. When launching, it says:
```

$ ./CarlaUE4.sh 
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
INTEL-MESA: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x4600041
  Serial number of failed request:  388
  Current serial number in output stream:  399
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
$
```

After doing `sudo sysctl dev.i915.perf_stream_paranoid=0`, the warning disappear but I still get segfault on both CARLA 0.9.11 and CARLA 0.9.10.1. 
```
$ ./CarlaUE4.sh
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x4600041
  Serial number of failed request:  388
  Current serial number in output stream:  399
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)
```

However it does work using CARLA 0.9.6 for example. 
```
$ ./CarlaUE4.sh
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
```

corkyw10, 2021-03-15:
@Axel1092 could you have a look at this please?

maria8899, 2021-03-15:
Hi, this is still happening. What's up with the intel MESA warning? is it because it is trying to launch on my intel CPU?

This laptop has a GeForce RTX 2060 and I am using Nvidia drivers 460.32.03 (which are working fine as used for my display) with CUDA 11.2 on Ubuntu 20.04 LTS.

Axel1092, 2021-03-16:
Hi @maria8899, I think it is as you suspect. It is trying to launch carla using the integrated intel graphics instead of the rtx 2060, causing carla to fail due to its high gpu demands. In the nvidia control panel you should be able to tweak some options to force the OS to use the nvidia gpu.

maria8899, 2021-03-16:
Actually it just works with the opengl flag as in `./CarlaUE4.sh -opengl`
Thank you!

closed 2021-03-16 by maria8899

HosseinTaherian, 2021-05-22:
Hello @maria8899 , I have same issue as you. Did you find any solution for that out of using -opengl?

maria8899, 2021-05-24:
@HosseinTaherian no, I have to use the opengl flag, this is related to vulkan not working well on Linux I think, check https://carla.readthedocs.io/en/latest/adv_rendering_options/#vulkan-vs-opengl

julienlau, 2021-10-07:
same issue here and nvidia-settings is set to NVIDIA only.
I do not want to blacklist intel driver, because I sometimes use it when needing battery saver mode.
I think Carla is grabbing the first active GPU device, instead of the one selected by PRIME. See screenshot below.

[image]

Using `-opengl` do not make things work, I get a message box and then the same error + coredump as without `-opengl`

[image]

How to check which GPU is using Carla ?

julienlau, 2021-10-07:
Linked to :
https://github.com/carla-simulator/carla/issues/3920
https://github.com/carla-simulator/carla/issues/3490
https://github.com/carla-simulator/carla/issues/3196
https://github.com/carla-simulator/carla/issues/2396

I don't think this should be played down, because it seems most Linux users using dual GPU have this issue

daniel-bogdoll, 2021-11-29:
I personally fix it with `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` which I found in `https://github.com/carla-simulator/carla/issues/2232`. Ubuntu 20.04 with RTX 3080 and Intel chip.

daniel-bogdoll, 2021-12-20:
@corkyw10 @Axel1092 should this issue be re-opened?

DogukanKuyumcu, 2022-03-10:
> I personally fix it with `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` which I found in #2232. Ubuntu 20.04 with RTX 3080 and Intel chip.

It fix the problem. Thanks

dkarunakaran, 2022-04-07:
> I personally fix it with `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` which I found in #2232. Ubuntu 20.04 with RTX 3080 and Intel chip.

Wow, this fixed my issue. I'm running Carla 0.9.13 on Ubuntu 20.04 with RTX 3060 and Intel processor

cchun319, 2022-04-10:
> I personally fix it with `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` which I found in #2232. Ubuntu 20.04 with RTX 3080 and Intel chip.

Great thanks.
It fixed mine. I am running Carla 0.9.13 on 20.04 with RTX 3060 and i7-11370H - ASUS TUF

ahv1373, 2023-09-26:
> I personally fix it with `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"` which I found in #2232. Ubuntu 20.04 with RTX 3080 and Intel chip.

Wonderful.
It fixed my problem