carla-simulator/carla #3922: map.get_waypoint (project_to_road=False) does not result in waypoints at expected locations

amzoo, 2021-02-24:
Carla 0.9.11 (package install using apt-get)
MacOS 11.2
Custom Map

---------------------------------

**API Call:** `map.get_waypoint(carla.Location(x, y, z), project_to_road=False, lane_type=carla.LaneType.Any)`
**Issue:** Attempting to generate waypoints off of lane center, at location (x, y, z) results in waypoints that are sometimes at the lane center, sometimes at location point, and sometimes somewhere in between. I have removed all None type waypoints from the resulting loop.

Number of input locations: 2203
Number of Waypoints (excluding NoneType): 1124

---------------------------------

**API Call:** `map.get_waypoint(carla.Location(x, y, z), project_to_road=True, lane_type=carla.LaneType.Any)`
**Result:** Works as intended, waypoints are generated on lane center, nearest to input location.

Number of input locations: 2203
Number of Waypoints (excluding NoneType): 2203

---------------------------------

**API Call:** `map.generate_waypoints(10)`
**Result:** Works as intended, waypoints are generated on lane center, equally spaced.

Number of Waypoints: 444

---------------------------------

Image below shows plots of 
blue: `map.generate_waypoints(10)`
red: `map.get_waypoint(carla.Location(x, y, z), project_to_road=False, lane_type=carla.LaneType.Any)`
green: `carla.Location(x, y, z)`

[image]
[image]

Axel1092, 2021-02-25:
Hi @amzoo, I think there is a misconception on how the `get_waypoint`function works. Waypoints are always objects with the position in the center of the lane. 
When you call the function `map.get_waypoint(carla.Location(x, y, z), project_to_road=False, lane_type=carla.LaneType.Any)` you are getting a waypoint corresponding to the lane (x, y, z) is in. If (x, y, z) is outside the lane the function will return `None`.
Finally, `generate_waypoints()` creates waypoints only in lanes of type driving, while your function may return a waypoint of any type because of the `carla.LaneType.Any`. So probably you are getting waypoints of sidewalks or other lane types. I suggest you try `lane_type=carla.LaneType.Driving` to get all the waypoints of the same type.

amzoo, 2021-02-26:
Understood! I realize now that those rogue waypoints I was seeing are actually on the center of the sidewalk. I misunderstood that the (x,y,z) was only being used to choose a specific lane.

Is there anyway to generate a waypoint not on the center of a lane? My use case is in the racing world, in which vehicles may travel anywhere on the drivable lane, and not stick to center.

Axel1092, 2021-03-03:
@amzoo Waypoints are representations of the road so their transform only points towards the center of the lane. You can however obtain displaced locations in the road using the `transform.get_right_vector()` to obtain a location laterally displaced from the center of the lane.

amzoo, 2021-03-03:
I can use carla.Locations fine, but I was hoping there were carla.Waypoints that I could move from the center of the lane in order to use the methods in the Waypoint class

Axel1092, 2021-03-15:
@amzoo, waypoints represent the center of the lane only. You need to need to do the displacement yourself

closed 2023-06-02 by stale[bot]