carla-simulator/carla #3925: fatal: Not a git repository (or any of the parent directories): .git

whuxhzhang, 2021-02-25:
ubuntu18.04
carla-0.9.11

I down load the source code of carla-0.9.11, run "make package" and find these errors:

fatal: Not a git repository (or any of the parent directories): .git
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 128

glopezdiest, 2021-02-25:
Hey, that generally means that you are running that command in a folder that isn't part of any git repository. Can you make sure that you are inside the CARLA folder and run the `make package` again?

glopezdiest, 2021-11-24:
As this seems solved, I'm closing the issue. Reopen it if needed

closed 2021-11-24 by glopezdiest

Li-Yongcheng, 2022-03-03:
ubuntu18.04
carla-0.9.13

I run `make PythonAPI`  in the root CARLA folder, but this error still comes out. I've no idea why this happens.

Tomato1107, 2022-07-20:
change get_git_repository_version to "0.9.11" will work

Johanfanas, 2022-10-05:
I changed get_git_repository_version to "0.9.11" and got this: /carla-0.9.11/Util/BuildTools/Setup.sh: line 468: 0.9.11: command not found.

PooriyaTT, 2022-12-09:
> I changed get_git_repository_version to "0.9.11" and got this: /carla-0.9.11/Util/BuildTools/Setup.sh: line 468: 0.9.11: command not found.

Could you please let me know how did you fix this problem? I am also getting this error after changing get_git_repository_version to 0.9.13.

azzamh15, 2024-04-04:
Can anybody explain how is this supposed to work?

> CARLA_VERSION=$(get_git_repository_version)

link422, 2024-06-11:
> Can anybody please explain how is this supposed to work?
> 
> > CARLA_VERSION=$(get_git_repository_version)

You need to change everything to the right of the equal sign. Since I am using 0.9.15 The line now reads: 
CARLA_VERSION="0.9.15"