carla-simulator/carla #3932: Physics values, distances, and more !

AIDRI, 2021-02-27:
Hello, 
I would like to know if there are physical values that are applied to simulators (I guess, yes), and what are these values, such as the gravity used, the mass of a vehicle, etc...
Also, can we calculate distances simply in carla simulator? Notably the distances in a curve.

The goal is to try to optimize the exit speed and the time spent in the turn.

Thank you in advance for your answers.

PS: is there any documentation relating to the physics and properties of the vehicles? I did not find any.

DSantosO, 2021-03-16:
Hello @AIDRI,
I don't think we have nothing to compute that but you can use the information of the [Waypoint API](https://carla.readthedocs.io/en/latest/python_api/#carla.Waypoint) to compute it.
You can control and change the physical properties of vehicles with the [Vehicle Physics Control](https://carla.readthedocs.io/en/latest/tuto_G_control_vehicle_physics/).

AIDRI, 2021-03-16:
Hi,
Thanks for your answer. I just check the Vehicule Physics Control part, and I think some vars may be interested.

closed 2021-03-30 by DSantosO