carla-simulator/carla #3933: make launch error Windows 10 carla 0.9.11 "The filename, directory name, or volume label syntax is incorrect."

jfolden, 2021-02-27:
Hi all, when ever I try and run **make launch** I'm hit with the same error, _"The filename, directory name, or volume label syntax is incorrect."_. I've tried everything in the FAQ, including deleting and re-cloning carla. The UE4_ROOT environment variable is pointing to my UE4 folder, and I have the carla content in "Content\carla". 

Here's the output from when I run **make launch**.
```
F:\Research\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  24
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "F:\Research\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "F:\Research\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "F:\Research\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "F:\Research\carla\Build\" --zlib-install-dir "F:\Research\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "F:\Research\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "F:\Research\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "F:\Research\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "F:\Research\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "F:\Research\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "F:\Research\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "F:\Research\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "F:\Research\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 24
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "F:\Research\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "F:\Research\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "F:\Research\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in F:\Research\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.11:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz

   Unzip it in the "F:\Research\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in F:\Research\carla\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server --client
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: F:/Research/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  carla_server.vcxproj -> F:\Research\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> F:\Research\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18363.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: F:/Research/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  carla_client.vcxproj -> F:\Research\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client.lib
  carla_client_debug.vcxproj -> F:\Research\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "F:\Research\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "F:\Research\carla\PythonAPI\carla\dependencies\"!
-[BuildCarlaUE4]: [Batch params]: --build --all
--build
--all
ECHO is off.
-[BuildCarlaUE4]: Building Unreal Editor...
The filename, directory name, or volume label syntax is incorrect.
make: *** [CarlaUE4Editor] Error 1

F:\Research\carla>
```
I've looked up that error for Windows, it may have something to do with an invalid character for a directory/file.

Any help with this issue would be greatly appreciated!

[image]

Zhuweilong123, 2021-02-28:
@jfolden 
Do you add UE4_ROOT(my path is D:\CARLA\CARLA_Latest\carla) to your environment variable?  UE4_ROOT should be located to the root directory of CARLA simulator.
[image]

If you set your environment variable correctly. I strongly recommend you to check the execution logic in BuildCarlaUE4.bat(my path is D:\CARLA\CARLA_Latest\carla\Util\BuildTools)
[image]

corkyw10, 2021-03-15:
@bernatx could you follow up on this please?

jfolden, 2021-03-15:
@corkyw10 I got it working. My specific problem was that Windows added a ';' to the end of my UE4_ROOT environment variable, which caused the path to be "C:\Path\to\UnrealEngine;" instead of "C:\Path\to\UnrealEngine".

What helped me was printing out the paths used in BuildCarlaUE4.bat (my path is D:\CARLA\CARLA_Latest\carla\Util\BuildTools).

Also, just running the .uproject in the Unreal folder in the carla directory helped out a ton, not with this problem but in general lol.

corkyw10, 2021-03-16:
@jfolden That's great, I'm glad you got it working!

closed 2021-03-16 by corkyw10