carla-simulator/carla #3939: C2665 :  none of the 2 overloads could convert all the argument types. Occurred while create a sensor

JeongKyuho, 2021-03-02:
I need help to create a sensor. I added a sensor by referring to the document, but I don't know why there is an error in the rebuild process. 

  SafeDistanceSerializer.cpp
c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(22): error C2665: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array': none of the 2 overloads could convert all the argument types. [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj]
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\array.h(148): note: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array(const carla::sensor::data::Array<carla::rpc::ActorId> &)'? ? ????.
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\array.h(136): note: ??       'carla::sensor::data::Array<carla::rpc::ActorId>::Array(size_t,carla::sensor::RawData &&)'
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(21): note: ?? ?? '(carla::sensor::RawData)'?(?) ????? ??
Done Building Project "C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj" (default targets) -- FAILED.
Project "C:\Projects\UE4\carla\Build\libcarla-visualstudio\ALL_BUILD.vcxproj" (3) is building "C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj" (5) on node 1 (default targets).
InitializeBuildStatus:
  Touching "carla_client_debug.dir\Release\carla_cl.D1D85111.tlog\unsuccessfulbuild".
CustomBuild:
  All outputs are up-to-date.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\CL.exe /c /IC:\Projects\UE4\carla\LibCarla\cmake\..\source /I"C:\Projects\UE4\carla\LibCarla\cmake\..\source\third-party" /I"c:\Projects\UE4\carla\Build\boost-1.72.0-install\include" /I"c:\Projects\UE4\carla\Build\rpclib-install\include" /I"c:\Projects\UE4\carla\Build\recast-cdce4e-install\include" /I"c:\Projects\UE4\carla\Build\libpng-1.2.37-install\include" /Zi /nologo /W3 /WX- /diagnostics:classic /MP /Od /Ob0 /D WIN32 /D _WINDOWS /D BOOST_ASIO_ENABLE_BUFFER_DEBUGGING /D _WIN32_WINNT=0x0600 /D HAVE_SNPRINTF /D BOOST_ERROR_CODE_HEADER_ONLY /D LIBCARLA_IMAGE_WITH_PNG_SUPPORT /D "CMAKE_INTDIR=\"Release\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /std:c++14 /Fo"carla_client_debug.dir\Release\\" /Fd"carla_client_debug.dir\Release\carla_client_debug.pdb" /Gd /TP /FC /errorReport:queue C:\Projects\UE4\carla\LibCarla\source\carla\sensor\s11n\SafeDistanceSerializer.cpp
  SafeDistanceSerializer.cpp
c:\projects\ue4\carla\build\boost-1.72.0-install\include\boost\asio\buffer.hpp(34): warning C4005: 'BOOST_ASIO_ENABLE_BUFFER_DEBUGGING': ??? ??? [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  c:\projects\ue4\carla\libcarla\source\carla\sensor\s11n\safedistanceserializer.cpp: note: 'BOOST_ASIO_ENABLE_BUFFER_DEBUGGING'? ?? ??? ??????.
c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(22): error C2665: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array':  none of the 2 overloads could convert all the argument types. [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\array.h(148): note: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array(const carla::sensor::data::Array<carla::rpc::ActorId> &)'? ? ????.
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\array.h(136): note: ??       'carla::sensor::data::Array<carla::rpc::ActorId>::Array(size_t,carla::sensor::RawData &&)'
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(21): note: ?? ?? '(carla::sensor::RawData)'?(?) ????? ??
Done Building Project "C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\Projects\UE4\carla\Build\libcarla-visualstudio\ALL_BUILD.vcxproj" (default targets) -- FAILED.
Done Building Project "C:\Projects\UE4\carla\Build\libcarla-visualstudio\install.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\Projects\UE4\carla\Build\libcarla-visualstudio\install.vcxproj" (default target) (1) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj" (default target) (5) ->
(ClCompile target) ->
  c:\projects\ue4\carla\build\boost-1.72.0-install\include\boost\asio\buffer.hpp(34): warning C4005: 'BOOST_ASIO_ENABLE_BUFFER_DEBUGGING': ??? ??? [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]

"C:\Projects\UE4\carla\Build\libcarla-visualstudio\install.vcxproj" (default target) (1) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj" (default target) (4) ->
(ClCompile target) ->
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(22): error C2665: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array':  none of the 2 overloads could convert all the argument types. [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj]

"C:\Projects\UE4\carla\Build\libcarla-visualstudio\install.vcxproj" (default target) (1) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj" (default target) (5) ->
  c:\projects\ue4\carla\libcarla\source\carla\sensor\data\safedistanceevent.h(22): error C2665: 'carla::sensor::data::Array<carla::rpc::ActorId>::Array':  none of the 2 overloads could convert all the argument types. [C:\Projects\UE4\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]

    1 Warning(s)
    2 Error(s)

this is my error message and i need help for solve the problem

JeongKyuho, 2021-03-02:
my software versions : 

carla 0.9.11-4
OS: windows 10
VS : 2017
Python : 3.7.9

corkyw10, 2021-03-16:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-16:
Hi @JeongKyuho, without seeing your changes I cannot know why are you getting these compilation errors. Apparently, your re misusing the `carla::sensor::data::Arraycarla::rpc::ActorId::Array` class/struct/function in your code but cannot give you any more insight without checking the code.

ZhanYangFan, 2021-04-26:
hello! boy, I miss the same problem,how do you solve the question.please help me! very thank you!

uservinu, 2021-11-09:
@JeongKyuho  @ZhanYangFan or anyone having these issues like me. You are following an old tutorial and it will not work with carla version 0.9.12 since the c++ APIs are updated. I have not tested any other versions. 
I managed to get the sensor working on 0.9.12 with the following changes.

SafeDistanceEvent.h
`#pragma once

#include "carla/rpc/ActorId.h"
#include "carla/sensor/data/Array.h"

namespace carla {
namespace sensor {
namespace data {

  class SafeDistanceEvent : public Array<rpc::ActorId> {
  public:

    explicit SafeDistanceEvent(RawData &&data) 
      : Array<rpc::ActorId>(sizeof(rpc::ActorId),std::move(data) ) //Array constructor is updated
        {

        }  
      }; 

} // namespace data
} // namespace sensor
} // namespace carla
 `