carla-simulator/carla #3943: Carla cannot run on a headless ubuntu driver

zxqzhang, 2021-03-03:
Hi,
I have rented a ubuntu server that cannot use docker,  so I installed Carla following the official guidelines. After installing it and setting the environment variable SDL_VIDEODRIVER=offscreen, the carla simulator still doesn't run, and the output shows
 
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.

Please help.

PTorr, 2021-03-03:
Try:
env SDL_VIDEODRIVER=offscreen bash -x /home/carla/CarlaUE4.sh -carla-server -world-port=2000 -opengl -quality-level=Epic -NOSOUND

grazvydas, 2021-03-04:
We had similar problem with our Ubuntu server. 
Can you run `nvidia-smi ` inside container and show the output? If GPU is available and has proper drivers, then run `ldconfig -p | grep -E 'nvidia|cuda'`  and print the output.
 You may need to install something like `libnvidia-gl-{NVIDIA_DRIVER_VERSION` in your docker image.

PTorr, 2021-03-07:
Additionally, I think that what you are getting is just a warning.
My server has the same output and it runs just fine.

closed 2023-06-02 by stale[bot]