carla-simulator/carla #3945: [0.9.11] world.get_actors() empty in synchronous mode

raymanfx, 2021-03-03:
CARLA version: 0.9.11
OS: Fedora 33

I'm basically doing the following:
1. Start the server
2. Start a client and spawn a vehicle, set server to synchronous mode with 20 FPS:
```
client = carla.Client(args.host, args.port)
client.set_timeout(2.0)
..
carla_world = client.get_world()
settings = carla_world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 1.0 / 20
carla_world.apply_settings(settings)
```
3. Start another client, try to get the first client so I can attach some sensors:
```
client = carla.Client(args.host, args.port)
client.set_timeout(2.0)
..
carla_world = client.get_world()
actors = carla_world.get_actors()
```

In step 3, the list of actors returned by `get_actors()` is empty in synchronous mode. When running in async mode (CARLA default), everything is fine. This does not appear to be documented - is it a bug?

FYI the same issue was mentioned in https://github.com/carla-simulator/carla/issues/2045 I believe.

jackbart94, 2021-03-03:
If you're in synchronous_mode, you must tick the World in the first client. Do you do that anywhere in your code?

raymanfx, 2021-03-03:
Yes, I'm doing it in the first client. My agent can drive the hero vehicle just fine, so server <-> client synchrony works fine. I can probably setup a minimal example for you to reproduce this if needed.

jackbart94, 2021-03-04:
> Yes, I'm doing it in the first client. My agent can drive the hero vehicle just fine, so server <-> client synchrony works fine. I can probably setup a minimal example for you to reproduce this if needed.

Please do, we definitely need more information to debug this.

raymanfx, 2021-03-17:
Okay, I found the issue.
Apparently this snippet: `carla_world = client.get_world()` is not executed in a blocking fashion even when synchronous mode is activated. I use the following workaround to wait until actors are populated:

```
        retry_count = 10
        while retry_count > 0:
            if len(carla_world.get_actors()) > 0:
                break
            else:
                time.sleep(1.0)

        if retry_count == 0 and len(carla_world.get_actors()) == 0:
            raise ValueError('Actors not populated in time')
```

Without this workaround, the actor list is empty in approx. 50% of the time for me.
This issue occurs even if I apply the sync mode before requesting the world via `carla_world = client.get_world()`. Perhaps it makes sense to change the implementation so the call blocks when synchronous mode is activated?

umateusz, 2021-04-29:
I ran into the same problem. I think the actor list is not populated before the first tick after client initialization. For me:
```
        client = carla.Client(CARLA_HOST, CARLA_PORT)
        client.set_timeout(CARLA_TIMEOUT)
        client.get_world().wait_for_tick()
```
solves the problem, but I think this is not the correct client behavior.

closed 2021-11-18 by jackbart94