carla-simulator/carla #3952: Error while trying examples

Mano2610, 2021-03-04:
(WindowsNoEditor) D:\Carla\WindowsNoEditor\PythonAPI\examples>python spawn_npc.py

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 301, in <module>
    main()
  File "spawn_npc.py", line 109, in main
    world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

I am facing this issue and couldn't find the port connection. Kindly help me to resolve this issue.

Mano2610, 2021-03-04:
I have also tried with python spawn_npc.py -n 80

But again getting the same error

ar4development, 2021-03-12:
@Mano2610 Couple of times I was experiencing the same issue. I think that could be related to the overall performance of your system. Can you try to run Carla with `-quality-level=Low` key?

Mano2610, 2021-03-12:
@ar4development - Do I need to run Carla4UE.exe-quality-level=low, while running the Carla environment.

Mano2610, 2021-03-12:
I have tried, but again getting the same error as below,

(WindowsNoEditor) C:\Users\kvman\Downloads\WindowsNoEditor>CarlaUE4.exe -quality-level=Low

(WindowsNoEditor) C:\Users\kvman\Downloads\WindowsNoEditor>cd PythonAPI\examples

(WindowsNoEditor) C:\Users\kvman\Downloads\WindowsNoEditor\PythonAPI\examples>python spawn_npc.py

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 301, in <module>
    main()
  File "spawn_npc.py", line 109, in main
    world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

ar4development, 2021-03-12:
@Mano2610 does your simulator look responsive before you try to spawn the actors? I mean can you "fly" over the map using w/a/s/d/e/q ?

Mano2610, 2021-03-12:
Yes, I am able to move with my keys w,a,s,d

corkyw10, 2021-03-16:
@Mano2610 are you running the examples on a CARLA map or custom? What version of CARLA are you using?

@bernatx could you have a look at this please?

Mano2610, 2021-03-16:
First I opened the Carla simulator and then I tried to run the example scripts like spwan_npc, manual_control_carsim, but I am unable to run those scripts. I am getting the runtime error like the simulator is not connected to the required port.

RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

I am using the latest version of Carla 0.9.11. I have also tried version 0.9.10, but no luck. I am getting the same error. 

Can you please help me to find the port connection of Carla?

closed 2023-06-02 by stale[bot]