carla-simulator/carla #3953: Some objects are not visible in LiDAR

bpoebiapl, 2021-03-05:
Environment: Windows 10 & Ubuntu 20.04
Version: 0.9.11
Map: Town10HD

Some objects (buildings, some kind of trees, traffic lights, ...) are not visible in LiDAR.
In certain situations, points may be taken through buildings or in places where the scanning does not sweep.

[image]
[image]

Note: color information acquired from rgb camera image added to the point cloud

corkyw10, 2021-03-16:
@DSantosO could you follow up on this please?

XGodina, 2021-03-17:
Hi @bpoebiapl,
We know what happens there with the lidar and collisions as well and now we are working to fix it

XGodina, 2021-03-19:
Hi @bpoebiapl , We've discovered the problems. Some Blueprint in our buildings, there are with a wrong collision preset. Now we add a correct collision preset and now appear in the lidar. The PR is waiting to accept.

bpoebiapl, 2021-08-20:
Has this issue resolved?

yuvalHG, 2021-09-12:
Is there any new with this issue? It was fixed on 0.9.12?