carla-simulator/carla #3960: Fixed time-step: How can I get desired fixed fps by world.setting ?

liuyandong1988, 2021-03-08:
When I use carla-linux-build version, I can make launch the carla. And set the available or fixed fps by UE4 edit.

> Edit->Project Settings -> Engine -> General Settings -> Framerate-> Use Fixed Frame Rate

[image]

And then I read the carla documentation. I want to modify fps in client-python-script.

> settings = world.get_settings() 
>
> settings.fixed_delta_seconds = 1.0/15
>
>  world.apply_settings(settings)

But I cannot get the result.  The fps seems to be running in a way that available fps.

[image]

when i cut the python script. UE4 editor fps is 

[image]

What mistake did I make? I want to control fps by client python script

DSantosO, 2021-03-16:
Hello @liuyandong1988,

The `settings.fixed_delta_seconds` sets a constant simulation time interval between frames. This is important to have a correct simulation but it has nothing to do with the actual framerate. Please check this [page](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep) for further documentation. 

From the client, that option of the engine is not exposed. We can consider adding it but in the meantime, I see two options. You could modify the source of your CARLA version and add these options following what we are doing for substepping for example. Another option is to control the framerate in the client with pygame for example and then, use `wait_for_tick()` if you are in asynchronous mode or `tick()` if you are in synchronous mode.

closed 2021-04-26 by DSantosO

liuyandong1988, 2021-05-13:
> Hello @liuyandong1988,
>
> The `settings.fixed_delta_seconds` sets a constant simulation time interval between frames. This is important to have a correct simulation but it has nothing to do with the [...]

Thank you for your reply. I can use the  `wait_for_tick()`  in asynchronous mode and make client and server have the same fps and use `tick()` in synchronous mode, server waits for the client. However, I can not fixed the server fps use the world.settings in [page](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep) . I want fixed the server fps...