carla-simulator/carla #3964: Carla client python wheel

psarka, 2021-03-08:
We are running carla server on the server and carla client on the client. As such, current packaging is a bit cumbersome, as python client requires the same ~8GB of space and only works on ubuntu, and requires a random egg matching python version.

It does not have to be this way - we managed to build a wheel that is cross distro is ~30MB. It can be installed properly with pip install [carla-client-unofficial](https://pypi.org/project/carla-client-unofficial/), and there is no need to manualy adjust the paths to find the .egg file in the client side scripts.

If you would be interested, this is a [commit](https://github.com/psarka/carla/commit/38c70fb2c1d6da8f8b58150e8297f8e851b467fa) that enables such a build - it is basically a Dockerfile based on manylinux2014 with a couple of minor changes. If one checks out that branch, the commands for building a wheel are:

    docker build -t carla-client-0.9.10.1-build -f CarlaClient.Dockerfile .
    docker run --rm -v $HOME/carla:/carla/ carla-client-0.9.10.1-build bash -c 'cd carla;PYTHON_VERSION=3.6 make PythonAPI'
    twine upload $HOME/carla/PythonAPI/carla/wheelhouse/

`PYTHON_VERSION` can be changed to build the wheel for 3.7 and for 3.8. Python 3.9 is currently not supported by your build scripts (some deprecated functions are used), but it should be able to fix it easily.

Cheers!

corkyw10, 2021-03-16:
@bernatx could you have a look at this please?

tiax615, 2021-04-21:
Hi psarka!
Im sorry. I cant find your wheel when use those command:
`pip install carla-client-unofficial`

tiax615, 2021-04-21:
> Hi psarka!
> Im sorry. I cant find your wheel when use those command:
> `pip install carla-client-unofficial`

:< Cant install in Windows

psarka, 2021-04-21:
Unfortunatelly the unofficial carla client is only available on Linux :(

Gr8ayu, 2021-04-26:
> Unfortunatelly the unofficial carla client is only available on Linux :(

is there any specific python version requirement for installing it on linux ? I am getting this while installing

```
$ pip3 install carla-client-unofficial
Collecting carla-client-unofficial
  Could not find a version that satisfies the requirement carla-client-unofficial (from versions: )
No matching distribution found for carla-client-unofficial

```

psarka, 2021-04-27:
Yes, python 3.6, 3.7 or 3.8.

qhaas, 2021-04-27:
Official support would be welcome (i.e. `make PythonAPI` defaulting to WHL), our HPC system administrator removed `easy_install` and thus `egg` support due to upstream deprecation.

qhaas, 2021-08-25:
CARLA 0.9.12 now packages as whl, thanks CARLA!

closed 2021-08-25 by psarka