carla-simulator/carla #3965: Carla rosbridge manual.py doesn't publishes control commands in autopilot mode

ImtiazUlHassan, 2021-03-09:
I'm trying to record the control commands i.e steer and throttle using Carla-Ros bridge. But when i launch Carla Ros bridge with ego vehicle example.Control commands are publish when  i drive manually. But when i set the auto-pilot on it no data is published on the topic.

when i set auto pilot off

[image]

when i set auto pilot on

[image]

is there a way i can publish command in auto pilot mode ?

joel-mb, 2021-03-16:
Hi @ImtiazUlHassan, actually this is the normal behaviour. When enabling the autopilot, the vehicle is controlled internally by the traffic manager so the vehicle commands are not published by the ROS bridge.
If you want to access that information you will have to publish it manually. Using the CARLA Python API you can get the last applied control via the `get_control` getter (https://carla.readthedocs.io/en/latest/python_api/#getters_16). **Take into account that the command you get is the one applied in the last frame.** There is no way to know the command that will be applied in the current frame without accessing the traffic manager code.

ImtiazUlHassan, 2021-03-18:
> Hi @ImtiazUlHassan, actually this is the normal behaviour. When enabling the autopilot, the vehicle is controlled internally by the traffic manager so the vehicle commands are not published by the [...]

I have figured it out. When we launch carla_ros_bridge_with_example_ego_vehicle it publishes a topic named ego_vehicle status. And in that topic there are all information available including throttle , steer , velocity etc

closed 2021-03-18 by ImtiazUlHassan