carla-simulator/carla #3968: spawn walker/ pedestrian

PTorr, 2021-03-09:
Hi,

I'm trying to spawn pedestrian at custom location.
Is there a way to force a walker to be spawned in location (x,y,z)?
If not is there a way to get the closest point on a side walk next to my ego vehicle? and then spawn the walker there?

Thank you in advance.

XGodina, 2021-03-15:
@PTorr 
Hi we use the traffic scenario to create customs situations. I think @glopezdiest could you help better than me.

glopezdiest, 2021-03-15:
Hey @PTorr.

> Is there a way to force a walker to be spawned in location (x,y,z)?

yes, and its the same as with a vehicle, which involves the use of [carla.World](https://carla.readthedocs.io/en/latest/python_api/#carlaworld)'s `try_spawn_actor`:

```py
walker_blueprint = "walker.pedestrian.0001" 
walker_transform = carla.Transform()  # Add the desired location and rotation
walker = world.try_spawn_actor(walker_blueprint, walker_transform)
```

> If not is there a way to get the closest point on a side walk next to my ego vehicle? and then spawn the walker there?

That's also possible with the [carla.Map](https://carla.readthedocs.io/en/latest/python_api/#carlamap) class.

```py
# Get the vehicle waypoint
vehicle_location = vehicle.get_location()
vehicle_waypoint = map.get_waypoint(vehicle_location)

# Move one lane to the right until the sidewalk is found
lane_type = None
while lane_type != carla.LaneType.Sidewalk:  
    vehicle_right_waypoint = vehicle_waypoint.get_right_lane()
    lane_type = vehicle_right_waypoint.lane_type

walker_transform = vehicle_waypoint.transform
walker_blueprint = "walker.pedestrian.0001" 
walker = world.try_spawn_actor(walker_blueprint, walker_transform)
```

PD: don't blindly trust this code as I haven't tested it, but any errors will be due to carelessness from my part, the code structure is still correct.

PTorr, 2021-03-16:
Thanks for the answers.

closed 2021-03-16 by glopezdiest