carla-simulator/carla #3976: Augmenting icons to windshield & creating Heads-up Display

erfpak7, 2021-03-11:
I would like to create a heads-up display by augmenting some of the captured information to the windshield (similar to attached pic). CARLA is already showing HUD on the left hand side but is not showing it on the driver's view and it's quite far from the reality. So I was wondering whether there's a way that we could project those information (velocity, distance to obstacle, etc.) to the windshield and add some appealing icons to it?

[image]

erfpak7, 2021-03-25:
Is there any update on this? Has anybody ever created such a heads-up feature?

XGodina, 2021-04-07:
Hi @erfpak7,
I know that UE4 provides some tools to customize your HUD. I advise you to start to familiarize yourself with that tools and later create a BluePrint to connect Hud with the information that you want.

erfpak7, 2021-04-20:
Hi @XGodina,
Thanks for your recommendation. Sure thing, I'll work on it ;)

XGodina, 2021-06-30:
Hi @erfpak7,

Any progress ?. I interested in that.

Mikooman, 2023-05-23:
Hey everyone, 
anyone ever implemented a custom hud like mentioned in the original question? 
I would also be interested in it.

closed 2023-06-02 by stale[bot]