carla-simulator/carla #3977: LaneMap geometry generated with errors when running RSS example against custom map

ar4development, 2021-03-12:
Hi,

I'm trying to run `manual_control_rss` against my custom [map](https://github.com/carla-simulator/carla/files/6129723/map.zip) that I have exported  from OSM and imported to simulator using `config.py` script.

However what I'm getting is the error message saying:

```
[2021-03-12 14:22:03.561] [ad_map_access] [warning] LaneMap geometry generated with errors
[2021-03-12 14:22:03.561] [error] withinValidInputRange(::ad::map::point::Longitude)>> 1070.04 out of valid input range [-180, 180]
[2021-03-12 14:22:03.561] [error] withinValidInputRange(::ad::map::point::GeoPoint)>> GeoPoint(longitude:1070.04,latitude:205.332,altitude:0) has invalid member
[2021-03-12 14:22:03.561] [ad_map_access] [error] Cannot convert from Geo to ECEF: Input point invalid.
Traceback (most recent call last):
  File "manual_control_rss.py", line 878, in <module>
    main()
  File "manual_control_rss.py", line 870, in main
    game_loop(args)
  File "manual_control_rss.py", line 787, in game_loop
    world = World(client.get_world(), args)
  File "manual_control_rss.py", line 155, in __init__
    self.restart()
  File "manual_control_rss.py", line 227, in restart
    self.rss_sensor = RssSensor(self.player, self.world,
  File "/mnt/old/home/alexey/Downloads/CARLA_0.9.11_RSS/PythonAPI/examples/rss/rss_sensor.py", line 110, in __init__
    self.sensor.listen(self._on_rss_response)
ValueError: Cannot convert from Geo to ECEF: Input point invalid.
Segmentation fault (core dumped)
```

Is there anything I need to fix in my map or in example code?

Thanks!

corkyw10, 2021-03-16:
@berndgassmann could you follow up on this please?

berndgassmann, 2021-03-16:
Hi,
at least I was able to reproduce the issue. The problem with loading the map with the RSS sensor is related to the missing geo-reference within the open drive file. Also CARLA server output shows that issue:
"WARNING: cannot parse georeference: ''. Using default values."

When checking the xodr file written by the server at CarlaUE4/Content/Carla/Maps/OpenDrive/OpenDriveMap.xodr
<?xml version="1.0" encoding="utf-8"?>
<OpenDRIVE>
    <header revMajor="1" revMinor="4" name="" version="1.00" date="Tue Mar 16 18:35:47 2021" north="893.91" south="0.00" east="2227.78" west="0.00"/>
    <road name="2-￐ﾹ ￐ﾜ￑ﾃ￑ﾀ￐ﾸ￐ﾽ￑ﾁ￐ﾺ￐ﾸ￐ﾹ ￐﾿￑ﾀ￐ﾾ￑ﾁ￐﾿￐ﾵ￐ﾺ￑ﾂ" length="1151.51017679" id="360" junction="-1">
        <link>
            <predecessor elementType="junction" elementId="1"/>
            <successor elementType="junction" elementId="2"/>
        </link>
....

one sees that the geo-reference tag is missing (see e.g. the CALRA built-in maps:)
        <geoReference><![CDATA[+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>

which defines the projection and respective geoReference.
As a workaround for this I've just put the above geoReference tag into the stored file CarlaUE4/Content/Carla/Maps/OpenDrive/OpenDriveMap.xodr becoming:

  <header revMajor="1" revMinor="4" name="" version="1.00" date="Tue Mar 16 18:35:47 2021" north="893.91" south="0.00" east="2227.78" west="0.00">

        <geoReference><![CDATA[+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>
      </header>

Then, at least the RssSensor is starting up.

The question to raise is if that osm imported map is actually useful for running RSS.
If I load that map into [QGIS](https://qgis.org/en/site/) with installed [ad-map-access plugin](https://github.com/carla-simulator/map/releases/tag/v2.4.3) the road layout looks not very well. Thing might work out at some of the "simple" intersection, but for sure not at the complex parts of the map, where the geometry could not have been reconstructed in a meaningful way.
To perform actual meaningful RSS evaluations, I'd suggest to make use of better map material.

Regards,
Bernd.

berndgassmann, 2021-03-16:
sorry, but the editor was not actually lucky with the plain xml snippets, I had added to the comment:

Original stored OpenDriveMap.xodr
```xml
<?xml version="1.0" encoding="utf-8"?>
<OpenDRIVE>
    <header revMajor="1" revMinor="4" name="" version="1.00" date="Tue Mar 16 18:35:47 2021" north="893.91" south="0.00" east="2227.78" west="0.00">
    <road name="2-￐ﾹ ￐ﾜ￑ﾃ￑ﾀ￐ﾸ￐ﾽ￑ﾁ￐ﾺ￐ﾸ￐ﾹ ￐﾿￑ﾀ￐ﾾ￑ﾁ￐﾿￐ﾵ￐ﾺ￑ﾂ" length="1151.51017679" id="360" junction="-1">
        <link>
```

manually edited
```xml
<?xml version="1.0" encoding="utf-8"?>
<OpenDRIVE>
      <header revMajor="1" revMinor="4" name="" version="1.00" date="Tue Mar 16 18:35:47 2021" north="893.91" south="0.00" east="2227.78" west="0.00"

        <geoReference><![CDATA[+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>
      </header>
    <road name="2-￐ﾹ ￐ﾜ￑ﾃ￑ﾀ￐ﾸ￐ﾽ￑ﾁ￐ﾺ￐ﾸ￐ﾹ ￐﾿￑ﾀ￐ﾾ￑ﾁ￐﾿￐ﾵ￐ﾺ￑ﾂ" length="1151.51017679" id="360" junction="-1">
        <link>

```

TigerStone93, 2022-11-15:
Is there any update on this problem?

I've encountered similar issue with a message `[info] ensureValid(::ad::physics::Distance)>> 10449777986291.096 value out of range

TigerStone93, 2022-11-15:
First, thank you for the fast response.

I've also found that the `LaneMap geometry` warning is not related with this shutdown problem.

I made my custom map by `RoadRunner` and import it into `CARLA 0.9.13` to run the RSS through `ScenarioRunner`.

But my python code popped up an error at the `self.sensor.listen(self._on_rss_response)` line with the `IndexError: Distance value out of range` and failed to load scenario in the custom map.

Eventually, your solution that adding a `<georeference>` line in `xodr` file worked perfectly.

Thank you for your thoughtful help!