carla-simulator/carla #3978: TypeError: unsupported operand type(s) for *: 'int' and 'ENUCoordinate'

ar4development, 2021-03-12:
1. Run Carla 0.9.11_RSS
2. Run `manual_control_rss.py`
3. In game window press `F2`

**OBSERVED in logs:**

Destroyed RSS sensor
  File "/mnt/old/home/alexey/Downloads/CARLA_0.9.11_RSS/PythonAPI/examples/rss/rss_sensor.py", line 412, in _on_rss_response
    self.debug_visualizer.tick(self.route, not response.proper_response.isSafe,
  File "/mnt/old/home/alexey/Downloads/CARLA_0.9.11_RSS/PythonAPI/examples/rss/rss_visualization.py", line 621, in tick
    self.visualize_route(dangerous, route)
  File "/mnt/old/home/alexey/Downloads/CARLA_0.9.11_RSS/PythonAPI/examples/rss/rss_visualization.py", line 648, in visualize_route
    self.visualize_enu_edge(edge, color, self._player.get_location().z)
  File "/mnt/old/home/alexey/Downloads/CARLA_0.9.11_RSS/PythonAPI/examples/rss/rss_visualization.py", line 663, in visualize_enu_edge
    carla_point = carla.Location(x=float(point.x), y=float(-1 * point.y), z=float(point.z + z_offset))
TypeError: unsupported operand type(s) for *: 'int' and 'ENUCoordinate'

**EXPECTED:**  Visualization enabled
**ENVIRONMENT:**

- Carla 0.9.11_RSS
- Ubuntu 20.04
- Python 3.8.5

ar4development, 2021-03-12:
Looks like I manager to overcome this issue by changing:
 
```
carla_point = carla.Location(x=float(point.x), y=float(-1 * point.y), z=float(point.z + z_offset))
```
to 

```
carla_point = carla.Location(x=float(point.x), y=float(-1 * float(point.y)), z=float(point.z + z_offset))
```
in `def visualize_enu_edge(self, edge, color, z_offset):` of `rss_visualization.py`..

However I am still not able to get those colored lines between ego and object as it was shown in demo.

corkyw10, 2021-03-16:
@fpasch could you follow up on this please?

varunjammula, 2022-06-03:
This is not fixed yet! I have the same issue.

M-Colley, 2026-10-24:
The reported arithmetic traceback is corrected in current `ue4-dev` at `401472072315fbd2e2b9b04974dce4c9304f3cda`. [PR #4604](https://github.com/carla-simulator/carla/pull/4604) merged the fix in commit `6fc187f4b38ba273dadd62824c3df09d2636156f`: [rss_visualization.py](https://github.com/carla-simulator/carla/blob/401472072315fbd2e2b9b04974dce4c9304f3cda/PythonAPI/examples/rss/rss_visualization.py) converts the coordinate components to floats before multiplication/addition.

I ran the actual historical 0.9.11 and current UE4 modules with float-convertible coordinate test doubles that lack multiplication. The historical module reproduces the exact TypeError; the current method produces the expected coordinates. This verifies that specific Python arithmetic error on this revision, without claiming a full RSS/server visualization test or a fix in every older packaged release.

closed 2026-11-13 by carla-triage[bot]