carla-simulator/carla #3980: AI-controlled walkers not spawning using spawn_npc.py with OpenDRIVE Standalone Mode

wgoforth, 2021-03-12:
I am unable to set destinations for AI-controlled walkers with the PythonAPI when using CARLA in OpenDRIVE standalone mode (config.py -x).

I'm running CARLA 0.9.11 on Windows 10 with Python 3.7 and the following example is on the Town05 map.

```
python37 spawn_npc.py -n 0 -w 1
Location(x=19.671911, y=-8.949682, z=1.000010)
Nav: failed loading binary
WARNING: NAV: Failed to set request to go to  40.2506 -11.0281 1.3528e-05
WARNING: NAV: failed to set max speed
spawned 0 vehicles and 1 walkers, press Ctrl+C to exit.
```

I modified the existing code in order to get a walker to spawn because using world.get_random_location_from_navigation() returns 'None'.
`        # -------------
        # Spawn Walkers
        # -------------
        # some settings
        percentagePedestriansRunning = 0.0      # how many pedestrians will run
        percentagePedestriansCrossing = 100.0     # how many pedestrians will walk through the road
        # 1. take all the random locations to spawn
        spawn_points = []
        odr_map = world.get_map()
        spawning_landmark = odr_map.get_all_landmarks_from_id(opendrive_id="2407")
        offset_location=carla.Location(spawning_landmark[0].transform.location.x + 1, spawning_landmark[0].transform.location.y + 1, spawning_landmark[0].transform.location.z + 1)
        for i in range(args.number_of_walkers):
            spawn_point = carla.Transform()
            #loc = world.get_random_location_from_navigation()
            loc = offset_location
            print(str(loc))
            if (loc != None):
                #spawn_point.location = loc
                spawn_point.location = loc
                spawn_points.append(spawn_point)`

However, the above error gets thrown when attempting to set the destination using this code.

`        world.set_pedestrians_cross_factor(percentagePedestriansCrossing)
        for i in range(0, len(all_id), 2):
            # start walker
            all_actors[i].start()
            # set walk to random point
            #all_actors[i].go_to_location(world.get_random_location_from_navigation())
            destination_landmark = odr_map.get_all_landmarks_from_id(opendrive_id="2406")
            
            all_actors[i].go_to_location(destination_landmark[0].transform.location)
            # max speed
            all_actors[i].set_max_speed(float(walker_speed[int(i/2)]))`

corkyw10, 2021-03-16:
@bernatx could you follow up on this please?

thytless, 2021-03-17:
`Nav: failed loading binary`
This error means that the walker navigation module cannot load data that saves navigation info. Pls make sure you have got all data downloaded. It should be under `Unreal/CarlaUE4/Content/Carla/Maps/Nav/`, you can also check it out with latest carla docker image.

bernatx, 2021-03-17:
Hi,

thytless is right.

It seems that the standalone module could not autogenerate the binary file needed for the pedestrians to navigate.
The main reason could be that the map is too big to be parsed by the RecastBuilder with current settings.
Is that the case?

wgoforth, 2021-03-17:
@bernatx Do I need to run CARLA from source in order to confirm that? Also, if the map were too big, would I expect the same behavior when using the UE4 version of the map or does the OpenDRIVE map have a larger memory footprint? I can spawn pedestrians on Town05 using the same script modification (targeting OpenDRIVE landmarks for spawn points) without a problem when I'm using the UE4 map.

closed 2023-06-02 by stale[bot]