carla-simulator/carla #3981: How to add trees in batch in CarlaUE4?

EchoYJJ, 2021-03-13:
Hi! I'm new to CarlaUE4 and not familiar with the basic operation.
I can only add trees one by one. If you  know how to add trees in batch or where to find the guidance, please give me advises.
Thanks in advance.
[image]

XGodina, 2021-03-15:
Hi @EchoYJJ 

In UE4 you can select the Option of Foliage. In there, there are some options that you can play and deploy many trees really easily.
[image]

EchoYJJ, 2021-03-27:
Hi! @XGodina 
Thanks a lot for your inspiration, and sorry for replying so late.
Except for the instrutions in UE4, I also learn from _https://www.youtube.com/watch?v=XYYfIYDqsDA_.
Two questions still confusing me, hope to receive your advices.

1.As a result of my fbx scene imported from Roadrunner not created in CARLA, the paint tool only shows up on what I create in CARLA(i.e. the buliding I drug from CARLA)
[image]
I'm wondering maybe I should add a invisible horizontal plain at z=0 to provide adding space.

2.The Foilage-paint tool seems to focus on the range, but I hope to paint the trees in stright line with fixed gap as is shown in Town 0X.
[image]

Hope to receive your views, thanks in advance!

closed 2023-06-02 by stale[bot]