carla-simulator/carla #3983: The UE4 window won't open?

cinjon, 2021-03-13:
I downloaded 0.9.11 on Ubuntu 18.04.5 and went through the simple quick start method.
The package came out fine s.t. I can run `./CarlaUE4.sh [-opengl]` in one terminal window and then `python spawn_npc.py -n 80` in another and both display text on the screen.

However, the first command does not open any UE4 window like expected and consequently the second command fails to animate. I just tested pygame alone and found that `pygame.display.list_modes()` says that "Video system not initialized". 

What's going on here? I'm on VNC and the display is clearly up and running.

cinjon, 2021-03-13:
Solved. This was because I was using gnu screen. :(

closed 2021-03-13 by cinjon

mtc-20, 2023-03-16:
> Solved. This was because I was using gnu screen. :(

Sorry, I am also having similar problems. Can you expand on what you meant by using gnu screen ? and how did you figure that out?