carla-simulator/carla #3993: dependencies/include\carla/Memory.h(9): fatal error C1083: Cannot open include file: 'boost/enable_shared_from_this.hpp': No such file or directory error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\amd64\\cl.exe' failed with exit status 2

xiaoyang666, 2021-03-17:
Ask for your help！！
windows 10 carla 0.9.9

dependencies/include\carla/Memory.h(9): fatal error C1083: Cannot open include file: 'boost/enable_shared_from_this.hpp': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\amd64\\cl.exe' failed with exit status 2

corkyw10, 2021-03-17:
Hi @xiaoyang666 could you please provide the complete output of the error? 
What command are you running?
Do you have multiple versions of Microsoft Visual Code installed? You should only have Visual Studio 2017 installed, otherwise there can be conflicts. You can see more information [here](https://carla.readthedocs.io/en/latest/build_windows/#visual-studio-2017)

xiaoyang666, 2021-03-17:
> Hi @xiaoyang666 could you please provide the complete output of the error?
> What command are you running?
> Do you have multiple versions of Microsoft Visual Code installed? You should only have Visual [...]

Thank you very much for your reply. When I run make pythonAPI command(win). Before this, many friends had the same problem. This is the file of my visual installation

closed 2023-06-02 by stale[bot]