carla-simulator/carla #3994: Carla client file descriptor issue

mrcoee, 2021-03-17:
**Environment**
This issue appears with carla version 0.9.11 on Ubuntu 18.04.

**Issue**

Hi team, 

I'm currently trying to randomly generate scenes to produce training data for a meta segmentation ai. Unfortunately, this produces an segmentation fault error after a few iterations. My workaround for this problem is to start carla as a subproccess, generate a scene, and terminate the subproccess afterwards. Hence, in every iteration I have to create a client which seemingly produces 8 file descriptors. These file descriptors are not cleaned up after I terminate carla and therefore accumulate. This eventually leads to an 'too many open file error' error in iteration 128 when the fd table of the os reaches its usually limit 1028 file descriptors. 

Is there a way to close a client and cleanup its file descriptors properly? By a quick search on this, I found the open issue #3197 which seems to be closely related to my problem as @hh0rva1h creates clients in a loop as well, which leads to the same problem.

hh0rva1h, 2021-11-29:
@bernatx Could you please have a look at this finally? Segfaults like this are frustrating for a lot of users.

bernatx, 2021-11-29:
Hi, sure, I will take a look

hh0rva1h, 2021-11-29:
@bernatx Thanks, quite recently we also have big troubles with https://github.com/carla-simulator/carla/issues/4935 which might be related.

varunjammula, 2022-01-30:
Any updates on this issue?

closed 2023-06-02 by stale[bot]