carla-simulator/carla #3995: Fog rendering distance and black sky (0.9.11)

TimLued, 2021-03-17:
Hello,

is it possible to change the fog rendering distance in 0.9.11?
The following image shows how it looks in Unreal by using
environment.py --fog 100 --fogfalloff 2 --fogdist 0

[image]

I want it more like it is shown in
https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#weather-setting
where the fog ist placed all over the map.

Also, can I switch off the darkening of the sky with fog levels over 40?

Best regards

Madecu, 2021-03-17:
Hello @TimLued,

To increase the rendering distance, you have to open BP_Sky, select Exponential Height Fog and increase "View Distance" in VolumetricFog category

[image]

In order to disable the decrease of the sun intensity over the amount of fog, you have to edit BP_Weather, "LightsIntensity" functions and skip the part that says "Subtract Sun Intensity From Fog Density"

[image]

I hope it helps

TimLued, 2021-03-17:
Thank you @Madecu
That's exactly what I wanted.

closed 2021-03-17 by TimLued