carla-simulator/carla #3996: Using _build_graph(), I can get a networkx graph. I want to delete specific line. Is there a good method?

Zhuweilong123, 2021-03-17:
Using _build_graph(), I can get a networkx graph. I want to delete specific line in networkx graph. Is there a good method?

```
world = client.load_world('Town01')
amap = world.get_map()
dao = GlobalRoutePlannerDAO(amap, 1)
grp = GlobalRoutePlanner(dao)
grp.setup()
graph, id_map, road_id_to_edge = grp._build_graph()
```

germanros1987, 2021-03-17:
Hi,

Do you mean you want to remove a segment from the graph? Could you please clarify?

Zhuweilong123, 2021-03-22:
In my project, there will be some congestion in some roads. So I want the ego-vehicle to avoid those congestion when ego-vehicle runs on a predetermined trajectory. Due to the route planning algorithm is based on networkx graph, so I want to disable the congested road in networkx graph. Now I have found a solution. Thanks for your reply. @germanros1987

closed 2021-03-22 by Zhuweilong123