carla-simulator/carla #3997: Carla image cook fail. AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)

xuisshoe, 2021-03-18:
I am trying to build a CARLA docker image. I have tried to build the CARLA image on fresh installed Ubuntu 16.04/18.04/20.04 but I encountered with this same error. 

ERROR: Cook failed.
       (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 25
The command '/bin/sh -c cd /home/ue4 &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   cd /home/ue4/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   make package &&   rm -r /home/ue4/carla/Dist' returned a non-zero code: 2

Due to the failure there is no container that I can connect to so I am yet able to figure out how to get access to the build log file. I have literally noting else in the system and I follow the procedure precisely. And idea what is causing the issue?

ebadi, 2021-04-06:
Exact same issue: 
```
...
  LogShaderCompilers: Display: Worker (9/13): shaders left to compile 8084
  LogShaderCompilers: Display: Worker (1/13): shaders left to compile 8074
  LogShaderCompilers: Display: Worker (10/13): shaders left to compile 8064
  LogShaderCompilers: Display: Worker (6/13): shaders left to compile 8054
Took 860.360585s to run UE4Editor, ExitCode=137
ERROR: Cook failed.
       (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 25
The command '/bin/sh -c cd /home/ue4 &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   cd /home/ue4/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   make package &&   rm -r /home/ue4/carla/Dist' returned a non-zero code: 2
```
There is no ` /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt` file in the container. The `**UnrealBuildTool**/Log.txt` file has no indication of error. 

```
ue4@7b896cc0b66a:~$ find . -name Log.txt
./UnrealEngine/Engine/Programs/UnrealBuildTool/Log.txt
ue4@7b896cc0b66a:~$ tail -f ./UnrealEngine/Engine/Programs/UnrealBuildTool/Log.txt
Timeline.Print: [ 1.781](+0.006) TargetMakefile.Save()
Timeline.Print: [ 1.789](+0.000) Reading dependency cache
Timeline.Print: [ 1.789](+0.000) Reading action history
Timeline.Print: [ 1.789](+0.001) ActionGraph.GetActionsToExecute()
Timeline.Print:   1.789          [ 0.000](+0.000) Prefetching include dependencies
Timeline.Print:   1.789          [ 0.000](+0.000) Cache outdated actions
Timeline.Print: [ 1.791](+106.668) ActionGraph.ExecuteActions()
Timeline.Print: [108.460](+0.029) <unknown>
Timeline.Print: [108.490](+0.000) FileMetadataPrefetch.Stop()
Timeline.Print: [108.491]
```

YWangXJ, 2021-05-03:
same issue. I tired to reinstall docker but the error is still there. Interesting thing I noticed last time is that my computer have freezed for couple of minutes, and then the error pops up. As shown in the figure below. 
[image]

Is this caused by the system freeze?  If so, how to prevent that? My hardware shouldn't be that bad.
[image]

YWangXJ, 2021-06-28:
Not sure if this should be the solution, but I solved this problem by adding the Swap of my system from 2G to 8G following this video. [](https://www.youtube.com/watch?v=Tteiw6MY4QI)

The "make package" process keeps saying something about "Missing cached shader map for material XXX" and adding up the number of shaders to be compiled, which drastically eats my memory and leads to system freeze.  My desktop has 15.6G memory and the "make package" definitely needs more than that.

In addition, I also encountered this issue #3758 after changing the system swap, which was fixed by following the recommended steps for dockerfile and .sh files

Hope it helps!

erouirior, 2021-06-30:
Even increase swap to 8G, I still encountered the same issue:
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed

Really frustrated, does someone know how to solve it?
Thanks a lot!

YWangXJ, 2021-06-30:
> Even increase swap to 8G, I still encountered the same issue:
> AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
> RunUAT ERROR: AutomationTool was unable to run successfully. [...]

did you encounter the error "Took XXXXXs to Run UE4Editor, Exitcode=137"? If yes, you are having the same issue with me. I still faced this error once after I increased the swap to 8G. After closing all programs running, it then works. So build it again and finger crossed. This docker build does need insanely high RAM.

erouirior, 2021-06-30:
OMG, lol, yes, the exitcode is 137. Thanks for your prompt reply. This really consume RAM...

closed 2023-06-02 by stale[bot]