carla-simulator/carla #3998: Initialization of current_state.velocity in PIDController.h

youngsend, 2021-03-18:
Hi, @jackbart94, 

When I read code related to PID controller used in traffic manager,
I find that in `StateUpdate` function, the `current_state.velocity` is initialized as `(current_velocity - target_velocity) / target_velocity`, which seems not to be a valid velocity. 
Could you tell me why `current_state.velocity` is initialized this way?

jackbart94, 2021-03-25:
Hi!

That variable probably doesn't have the best name. For our PID do not use the target velocity itself, rather we use the **error** from our current velocity and the target velocity we want to reach. Then, it's divided by the target velocity to make the error a bit smaller, which helps reducing the oscillations when the PID is close to 0.

It will work with current_velocity - target_velocity as well, we just chose that particular formula because we had some issues tuning our PID and this solution helped us. Ultimately, the goal of the PID is to get as close to 0 as possible, and we want to help as we can!

youngsend, 2021-03-26:
Thank you @jackbart94 for the clear explanation!

closed 2021-03-26 by youngsend