carla-simulator/carla #3999: XIO:  fatal IO error 2 (No such file or directory) on X server ":0"

abirbensoltane, 2021-03-19:
Hello! I have problem while trying to launch the server . 
I tried 0.9.9.4 and 0.9.11 both give the same error 

$  ./CarlaUE4.sh
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
XIO:  fatal IO error 2 (No such file or directory) on X server ":0"
      after 341 requests (339 known processed) with 162 events remaining.
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

abirbensoltane, 2021-03-21:
The cause of this problem is Nvidia driver because I have two graphic cards.Most of people have some issue with nvidia drivers under linux. This is frustrating . I can't open Xserver as a simple user under linux and I can't run ./CarlaUE4.sh as root as well so I switched to windows but I wish I could find a solution.

Axel1092, 2021-04-13:
@abirbensoltane, this looks indeed a driver issue. Make sure that your discrete GPU has sufficient memory (we recommend 6GB although 4GB should work most of the time). You need to make sure to use that GPU and not the small one.

abirbensoltane, 2021-04-14:
> @abirbensoltane, this looks indeed a driver issue. Make sure that your discrete GPU has sufficient memory (we recommend 6GB although 4GB should work most of the time). You need to make sure to use that GPU and not the small one.
Well I have GTX1650ti with 4 GO and the other one is amd but when I use the GTX  only the screen becomes dark that's why I think it's prob of driver

abirbensoltane, 2021-04-15:
> @Axel1092  I've finally managed to solve the prob after installing the latest update of nvidia drivers today. But carla doesn't launch the whole pc freeze when I launch it

closed 2021-06-30 by Axel1092