carla-simulator/carla #4000: How to run faster in the co-simulation between CARLA & SUMO?

EchoYJJ, 2021-03-20:
Hi! I'm trying to run more simulation time in real time. (e.g. 1s in real life →1 min in simulation world)
To achieve this, I've tried change the **settings.synchronous_mode** from TURE to FALSE in the wish to run the server as fast as possible, but it doesn't work.
If you know how to solve it, please give some advises, thanks in advance!

EchoYJJ, 2021-03-31:
Also, when my scene gets more complex(more vegetation or vehicles), the simulation speed slows down(1s in real life≈5s in simulation). I also notice that vegetation in CARLA are dynamic which is unnecessary for me. Whether the simulation speed will rise, if I change the mode from dynamic to static.

joel-mb, 2021-03-31:
Hi @EchoYJJ. The co-simulation between CARLA and SUMO needs to be run in synchronous mode for both simulators. Otherwise, it is likely to lose the synchronization between both simulations at some point.

I've made a couple of changes to get faster co-simulation between CARLA and SUMO by default in this branch: https://github.com/carla-simulator/carla/tree/joel-mb/sumo_layers. Could you test if this is speeding up the co-simulation for your use case? I've added a new argument to be able to unload all the map layers and increase the performance (https://carla.org/2020/12/22/release-0.9.11/#map-sublevels). Take into account that this feature is available in the latest CARLA version (0.9.11) and that you need to use `Opt` maps. 

```
# Load opt map for Town01 (execute in <CARLA_ROOT>/PythonAPI/util)
python3 config.py -m Town01_Opt

# Run co-simulation (execute in <CARLA_ROOT>/Co-Simulation/Sumo)
python3 run_synchronization.py examples/Town01.sumocfg --sumo-gui --unload-all-map-layers
```

Also, take into account that the `run_syncrhonization` script will be always faster than `spawn_npc_sumo`. The lastest needs to check in every tick if the SUMO vehicle has arrived to the end of its route and reassing a new one if needed.

EchoYJJ, 2021-04-02:
Thanks a lot for your replying!
I think the first thing I need to do is to update my CARLA 0.9.10 to 0.9.11.
Also, I first notice that github sources have branches from different authors. Can I perceive it as slight improvement to handle bugs but still base on CALRA 0.9.11, and you focus on the SUMO co-simulation?

joel-mb, 2021-04-06:
@EchoYJJ Yes, you need to use at least CARLA 0.9.11. You can download it from here: https://github.com/carla-simulator/carla/releases/tag/0.9.11. You can also build CARLA from source using the branch I said in my previous comment. However, for this test, is enough using the package.

We use different branches to introduce new features and bug fixes into the `dev` branch. In the end, the `dev` branch is the one used to release new versions of CARLA (e.g., 0.9.12).

closed 2023-06-02 by stale[bot]