carla-simulator/carla #4001: sensors took too long to send their data error

AIasd, 2021-03-20:
Hi I encounter an error `Error during the simulation: > A sensor took too long to send their data` recently when using Carla 0.9.10 and Carla 0.9.11 (However, the errors are a bit different). 

On Carla 0.9.9, I am not encountering this error.

On Carla 0.9.10, I am encountering this error every time I run a scenario.

On Carla 0.9.11, I am encountering this error only when I run more than ~50 scenarios consecutively using the same server or running three carla servers simutaneously on different port on the same machine.

I use the following sensors:

```
[
                {
                    'type': 'sensor.camera.rgb',
                    'x': 1.3, 'y': 0.0, 'z': 1.3,
                    'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
                    'width': 256, 'height': 144, 'fov': 90,
                    'id': 'rgb'
                    },
                {
                    'type': 'sensor.camera.rgb',
                    'x': 1.2, 'y': -0.25, 'z': 1.3,
                    'roll': 0.0, 'pitch': 0.0, 'yaw': -45.0,
                    'width': 256, 'height': 144, 'fov': 90,
                    'id': 'rgb_left'
                    },
                {
                    'type': 'sensor.camera.rgb',
                    'x': 1.2, 'y': 0.25, 'z': 1.3,
                    'roll': 0.0, 'pitch': 0.0, 'yaw': 45.0,
                    'width': 256, 'height': 144, 'fov': 90,
                    'id': 'rgb_right'
                    },
                {
                    'type': 'sensor.other.imu',
                    'x': 0.0, 'y': 0.0, 'z': 0.0,
                    'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
                    'sensor_tick': 0.05,
                    'id': 'imu'
                    },
                {
                    'type': 'sensor.other.gnss',
                    'x': 0.0, 'y': 0.0, 'z': 0.0,
                    'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
                    'sensor_tick': 0.01,
                    'id': 'gps'
                    },
                {
                    'type': 'sensor.speedometer',
                    'reading_frequency': 20,
                    'id': 'speed'
                    },
                {
                    'type': 'sensor.camera.semantic_segmentation',
                    'x': 0.0, 'y': 0.0, 'z': 100.0,
                    'roll': 0.0, 'pitch': -90.0, 'yaw': 0.0,
                    'width': 512, 'height': 512, 'fov': 5 * 10.0,
                    'id': 'map'
                    },
                {
                    'type': 'sensor.camera.rgb',
                    'x': -6, 'y': 0.0, 'z': 3,
                    'roll': 0.0, 'pitch': -20.0, 'yaw': 0.0,
                    'width': 256, 'height': 144, 'fov': 90,
                    'id': 'rgb_with_car'
                    }
                ]
```
After printing out what sensors time out from receiving sensor information the servers, I notice that they are `'imu', 'gps',  'map' and 'rgb_with_car'`. Other sensor information in contrast are passed instantly. I tried to increase `'sensor_tick' `as well as decrease fps and increase fixed delta of the simulations but these do not help.

Can anyone shed some light on the possible issues here? I have really been stuck on this error for a while.

corkyw10, 2021-03-29:
@DSantosO could you follow up on this please?

DSantosO, 2021-03-30:
Are you using synchronous or asynchronous mode? What timeout have you configured?

AIasd, 2021-03-30:
Hi @DSantosO , I am using the synchronous mode and I set timeout to be 60 seconds. However, even if I increase the timeout time, the same error appears.

glopezdiest, 2021-04-28:
Hey @AIasd, sorry for the delay. Could you specify which versions are you using of CARLA, SR and Leadeboard? Additionally, have you modified the Leaderboard? Because your sensor configuration is weird. Your cameras are fine but you added the `sensor_tick` to the IMU and GNSS, which aren't being parsed (check [here](https://github.com/carla-simulator/leaderboard/blob/master/leaderboard/autoagents/agent_wrapper.py#L77) to see which values are hardcoded and which you can modify). Also, the semantic segmentation camera is not an allowed sensor, which should be raising an error.

AIasd, 2021-04-29:
Hi @glopezdiest  , I used the [Learning By Cheating version of Leaderboard](https://github.com/bradyz/leaderboard) but I updated it according to the latest SR for 0.9.11. The agent_wrapper I am using from [here](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/leaderboard/autoagents/agent_wrapper.py) which supports the semantic segmentation camera. Also since `sensor_tick` is ignored, they should not cause a problem, right?

glopezdiest, 2021-04-29:
> Also since sensor_tick is ignored, they should not cause a problem, right?

Yeah, indeed, just wanted to make sure about it.

Some explanation, what the leaderboard does is create a queue where all the sensor data is palced, and at some point, that data is retrieved, and sent to the agent. If there are less elements than sensors, it means that a sensor hasn't sent their data yet, so it waits. This is done [here](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/leaderboard/envs/sensor_interface.py).

What I'd recommend is to uncomment the prints at lines [219](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/leaderboard/envs/sensor_interface.py#L219) and [237](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/leaderboard/envs/sensor_interface.py#L219), and check how the queue changes. I'm specially worried for the initialization, as you have changed something when upgrading to 0.9.11, causing the queue to start incorrectly. Theoretically, there's only one tick between the creation of the sensors and the first time we retrieve the data from the queue, but that might be wrong in your case.

If that isn't the case and indeed, the sensor isn't sending its data, then that's another problem

AIasd, 2021-04-30:
Hi @glopezdiest ,

I will give this a try and report the results back in a couple of days. Thanks in advance!

AIasd, 2021-05-03:
Hi @glopezdiest ,

The following is the output of a timeout after I uncommented the prints.

```
setup callback rgb
setup callback rgb_left
setup callback rgb_right
setup callback rgb_with_car
setup callback imu
setup callback gps
setup callback speed
setup callback map

Updating gps - 20
Updating imu - 20
Updating rgb_with_car - 20
Updating rgb_right - 20
Updating rgb_left - 20
Updating rgb - 20
> Running the route
start to run scenario
Getting gps - 20
Getting imu - 20
Getting rgb_with_car - 20
Getting rgb_right - 20
Getting rgb_left - 20
Getting rgb - 20

Updating speed - 20
Getting speed - 20
Watchdog exception - Timeout of 59.0 seconds occured

Error during the simulation:
> A sensor took too long to send their data
```

and this one is the output of a successful run.
```
setup callback rgb
setup callback rgb_left
setup callback rgb_right
setup callback rgb_with_car
setup callback imu
setup callback gps
setup callback speed
setup callback map

Updating gps - 20
Updating imu - 20
Updating map - 20
Updating rgb_with_car - 20
Updating rgb_right - 20
Updating rgb_left - 20
Updating rgb - 20
> Running the route
start to run scenario
Getting gps - 20
Getting imu - 20
Getting map - 20
Getting rgb_with_car - 20
Getting rgb_right - 20
Getting rgb_left - 20
Getting rgb - 20
Updating speed - 20
Getting speed - 20

Resetting ego-vehicle!
Updating gps - 21
Updating imu - 21
Updating map - 21
Updating rgb_with_car - 21
Updating rgb_right - 21
Updating rgb_left - 21
Updating rgb - 21
Getting gps - 21
Getting imu - 21
Getting map - 21
Getting rgb_with_car - 21
Getting rgb_right - 21
Getting rgb_left - 21
Getting rgb - 21
Updating speed - 21
Getting speed - 21
...
```

The program has successfully finished ~10 scenarios with output similar to the second block and then for the next scenario it outputs the output similar to the first block and times out.
The difference between the two output blocks in this case seems to be the `map` sensor was successfully updated in the second one but not in the first one and led to the timeout.

glopezdiest, 2021-05-04:
> The difference between the two output blocks in this case seems to be the sensor.camera.semantic_segmentation(with id map)
Exactly, it doesn't seem to be sent on the first frame, which causes it to trigger the timeout. Okay, I'll check how you are setting up the semantic camera and report back, but so far I haven't seen anything weird.

closed 2021-07-21 by AIasd

yuliangguo, 2023-09-25:
If this problem solved? I seems to encounter the same issue when running the 'capture-sensor_data.py' nowadays based on the instructions from https://leaderboard.carla.org/get_started/

Noce99, 2024-02-07:
Same for me!

milk-bottle-liyu, 2024-04-11:
Same problem for Leaderboard2.0 agent.