carla-simulator/carla #4003: 'sensor_tick' attribute is non functional for ray-cast sensors

yaadram, 2021-03-21:
Carla version : 0.9.11
OS: Windows 10 64
Python 3.7

After upgrading to 0.9.11 it seems that all the ray-cast sensors (Did not check others) have a bug that 'sensor_tick' attribute is non functional. The sensor tick-time is as the simulation FPS , verified also in synchronous-mode and asynchronous-mode.
**Used to work fine in 0.9.9.** 

Any workaround/fix for this ?

heethesh, 2021-03-28:
same issue here

qqqGpe, 2021-04-25:
Hi, I just meet the same problem, do you have any solution?

yaadram, 2021-04-25:
> Hi, I just meet the same problem, do you have any solution?

Ya, they fixed it in :[https://github.com/carla-simulator/carla/pull/3818]
You will need to rebuild Carla... A workaround is filtering frames arriving in Python...

qqqGpe, 2021-04-26:
> > Hi, I just meet the same problem, do you have any solution?
> 
> Ya, they fixed it in :[https://github.com/[/pull/3818](https://github.com/carla-simulator/carla/pull/3818)]
> You will need to rebuild Carla... A workaround is filtering frames arriving in Python...

thanks you very much!

closed 2023-06-02 by stale[bot]