carla-simulator/carla #4005: Struggling to create a custom map from existing map.

alpkumbasar, 2021-03-22:


yaadram, 2021-03-22:
Hi alpkumbasar,

What I usually do is:
1. Make a copy of the map in the UE editor (File-> Save current as)
2. Use the UE editor to delete specific objects,  make copies of specific objects , or add new objects.
3. What i couldnt find a soluciton to: Changing semantic data like lanes, and the traffic flow, also creating new road is very difficult , and you might need "RoadRunner" or similar 3rd party applications.

I hope this helps,
Yaad

alpkumbasar, 2021-03-22:
Thanks,
After changing the things i do not want, i try to use ./build.sh inside DockerUtil/dist and it gives me a new file called crosswalks.obj, However in docs it says it should give me .bin file and i should move it to Nav folder.

To sum up, problem is i do not have .bin file and i could not find Nav folder in carla repo. 

Again thanks :))

corkyw10, 2021-08-06:
Hi @alpkumbasar sorry for the delay in answering. We now have updated map generation documentation [here](https://carla.readthedocs.io/en/0.9.12/tuto_M_custom_map_overview/). Have a look, there's many different routes depending on how you are using CARLA. If you have any new issues, please open a new github issue.

For pedestrian generation, look [here](https://carla.readthedocs.io/en/0.9.12/tuto_M_generate_pedestrian_navigation/). Sometimes the problem is that not all the assets are selected properly when you export the `.obj`. Double check in the `.obj` file that is generated that the sidewalk meshes are present.

closed 2021-08-06 by corkyw10