carla-simulator/carla #4006: How do I spawn at z=0.0?

cinjon, 2021-03-22:
I am trying to save generated scenes from Carla and then respawn them exactly later. Sometimes, I might move an NPC in some direction, but trivially and still within the camera's view. However, I run into an issue where if I try to spawn at z=0.0, then there is a collision with the ground. How do I fix that? Is there a way to turn off the collision detection and allow for npcs to be in that position regardless?

cinjon, 2021-03-29:
I run into this issue an enormous amount. Is there no way to specify that I want to spawn on the ground rather than waiting for the NPC to come falling down afterward?

corkyw10, 2021-03-29:
@XGodina could you follow up on this please?

cinjon, 2021-03-29:
Thanks for triaging. The issue at large is that I have to drop the agents from up high and wait for them to settle before I can get a snapshot of the scene at rest. It would be great to avoid this because it's adding a lot of sim time.

cinjon, 2021-03-30:
A little bit more information: I am generating scenes by having the agents go on autopilot for a while and then getting the last position after, say, 100 steps. I'm then regenerating that last scene, with some NPCs possibly moved. Sometimes, I can't even regenerate the original scene without a collision, which means I have to increase the z at which the colliding agent must "drop". This is a huge bottleneck for me because, while I'm getting great FPS in Carla, the resulting FPS after having to run for 30-50 frames to "drop" the NPCs just to get one frame is really poor.

cinjon, 2021-03-31:
@corkyw10 Is it possible to get the NPCs to "drop" faster per frame? Here's an example of what happens when I let it run for only 20 steps:

[image]

cinjon, 2021-04-01:
Hi, is the answer given in #3788 around changing the Collisions Complexity applicable here as well? I am using a fully automated approach and do not access any gui (such as from RoadRunner).

XGodina, 2021-04-01:
Hi@cinjon, 
In all our maps there are complexity collisions. Regarding collisions we change it from UE4, I don' know if is useful for you. As for the rest maybe @bernatx  or @Axel1092  could you help.

cinjon, 2021-04-01:
I'm not sure I understand - are you saying that there is a way to toggle the Collisions being respected? I would prefer to have them reported but still perform the action. This would be of great help.

Axel1092, 2021-04-13:
Hi @cinjon, vehicles cannot be spawned if there is another object intersecting your spawn location (in this case the road). z=0 is usually the surface of the road, but in order to spawn a vehicle there you should use the waypoint API like this:
```
waypoint = carla_map.get_waypoint(location)
spawn_transform = waypoint.transform
# spawn your vehicle
```