carla-simulator/carla #4007: "make PythonAPI" failed

sissiok, 2021-03-22:
Carla version: 0.9.11 (source)
OS: Ubuntu 18.04.5 LTS
Python version: 3.7.6

`make PythonAPI` failed and here is the error message:
```
2021-03-22 16:06:28 (4.25 MB/s) - ‘boost_1_72_0.tar.gz’ saved [126580835/126580835]

Setup.sh: Extracting boost for Python 3.
Building Boost.Build engine with toolset clang... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1
```

I checked `bootstrap.log` and I copied it as below. I appreciate if someone can help me with this!  
```
###
###
### Using 'clang' toolset.
###
###
/usr/bin/clang++-8 --version
clang version 8.0.1- (branches/release_80)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
###
###
/usr/bin/clang++-8 -x c++ -O3 -std=c++11 -O3 -s -DNDEBUG builtins.cpp class.cpp command.cpp compile.cpp constants.cpp cwd.cpp debug.cpp debugger.cpp execcmd.cpp filesys.cpp frames.cpp function.cpp glob.cpp hash.cpp hcache.cpp hdrmacro.cpp headers.cpp jam.cpp jambase.cpp jamgram.cpp lists.cpp make.cpp make1.cpp md5.cpp mem.cpp modules.cpp native.cpp object.cpp option.cpp output.cpp parse.cpp pathsys.cpp regexp.cpp rules.cpp scan.cpp search.cpp strings.cpp subst.cpp sysinfo.cpp timestamp.cpp variable.cpp w32_getreg.cpp modules/order.cpp modules/path.cpp modules/property-set.cpp modules/regex.cpp modules/sequence.cpp modules/set.cpp execunix.cpp fileunix.cpp pathunix.cpp -o b2
/tmp/class-fd9c53.o: In function `make_class_module(_list*, _list*, frame*)':
class.cpp:(.text+0x25): undefined reference to `string_new(string*)'
class.cpp:(.text+0x32): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x3d): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x52): undefined reference to `string_free(string*)'
class.cpp:(.text+0x197): undefined reference to `string_new(string*)'
class.cpp:(.text+0x1a4): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x1af): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x1c4): undefined reference to `string_free(string*)'
/tmp/class-fd9c53.o: In function `import_base_rule(void*, void*)':
class.cpp:(.text+0x2c8): undefined reference to `string_new(string*)'
class.cpp:(.text+0x2d3): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x2e0): undefined reference to `string_push_back(string*, char)'
class.cpp:(.text+0x2eb): undefined reference to `string_append(string*, char const*)'
class.cpp:(.text+0x2ff): undefined reference to `string_free(string*)'
/tmp/debugger-ef5192.o: In function `debug_end_stepping()':
debugger.cpp:(.text+0x150): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x15d): undefined reference to `string_reserve(string*, unsigned long)'
debugger.cpp:(.text+0x187): undefined reference to `string_push_back(string*, char)'
debugger.cpp:(.text+0x1ae): undefined reference to `string_free(string*)'
/tmp/debugger-ef5192.o: In function `handle_line_breakpoint(_object*, int)':
debugger.cpp:(.text+0x2a9): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x2ce): undefined reference to `string_free(string*)'
debugger.cpp:(.text+0x2fc): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x31f): undefined reference to `string_free(string*)'
debugger.cpp:(.text+0x369): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x38c): undefined reference to `string_free(string*)'
debugger.cpp:(.text+0x3b7): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x3da): undefined reference to `string_free(string*)'
/tmp/debugger-ef5192.o: In function `debug_on_breakpoint(int)':
debugger.cpp:(.text+0x510): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x51d): undefined reference to `string_reserve(string*, unsigned long)'
debugger.cpp:(.text+0x547): undefined reference to `string_push_back(string*, char)'
debugger.cpp:(.text+0x56e): undefined reference to `string_free(string*)'
/tmp/debugger-ef5192.o: In function `debugger()':
debugger.cpp:(.text+0x8dd): undefined reference to `string_free(string*)'
debugger.cpp:(.text+0x906): undefined reference to `string_new(string*)'
debugger.cpp:(.text+0x913): undefined reference to `string_reserve(string*, unsigned long)'
debugger.cpp:(.text+0x937): undefined reference to `string_push_back(string*, char)'
[... 77 lines cut ...]
debugger.cpp:(.text+0x4685): undefined reference to `string_free(string*)'
/tmp/property-set-ada739.o: In function `property_set_create(frame*, int)':
property-set.cpp:(.text+0x474): undefined reference to `string_new(string*)'
property-set.cpp:(.text+0x481): undefined reference to `string_append(string*, char const*)'
property-set.cpp:(.text+0x48c): undefined reference to `string_append(string*, char const*)'
property-set.cpp:(.text+0x499): undefined reference to `string_append(string*, char const*)'
property-set.cpp:(.text+0x4d4): undefined reference to `string_free(string*)'
/tmp/regex-66e026.o: In function `regex_replace(frame*, int)':
regex.cpp:(.text+0xe7): undefined reference to `string_new(string*)'
regex.cpp:(.text+0x10a): undefined reference to `string_append_range(string*, char const*, char const*)'
regex.cpp:(.text+0x115): undefined reference to `string_append(string*, char const*)'
regex.cpp:(.text+0x136): undefined reference to `string_push_back(string*, char)'
regex.cpp:(.text+0x15e): undefined reference to `string_append(string*, char const*)'
regex.cpp:(.text+0x17a): undefined reference to `string_free(string*)'
/tmp/regex-66e026.o: In function `regex_transform(frame*, int)':
regex.cpp:(.text+0x2a3): undefined reference to `string_new(string*)'
regex.cpp:(.text+0x307): undefined reference to `string_append_range(string*, char const*, char const*)'
regex.cpp:(.text+0x329): undefined reference to `string_truncate(string*, unsigned long)'
regex.cpp:(.text+0x36b): undefined reference to `string_free(string*)'
/tmp/pathunix-a6886a.o: In function `path_get_temp_path_(string*)':
pathunix.cpp:(.text+0x2f): undefined reference to `string_append(string*, char const*)'
/tmp/pathunix-a6886a.o: In function `path_translate_to_os_(char const*, string*)':
pathunix.cpp:(.text+0x4b): undefined reference to `string_copy(string*, char const*)'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

sissiok, 2021-03-23:
Update: 
`make PythonAPI` succeeded by downgrading boost from version 1.72.0 to 1.69.0 https://github.com/carla-simulator/carla/blob/78e7ea11306ca164fb664ec74d2224f2e1d01923/Util/BuildTools/Setup.sh#L104

closed 2021-03-23 by sissiok

BSRKLS, 2024-08-20:
how did you manage to downgrade it?

BSRKLS, 2024-08-20:
> Update: `make PythonAPI` succeeded by downgrading boost from version 1.72.0 to 1.69.0
> 
> https://github.com/carla-simulator/carla/blob/78e7ea11306ca164fb664ec74d2224f2e1d01923/Util/BuildTools/Setup.sh#L104

how have you achieved to downgrade it?