carla-simulator/carla #4009: Car light brightness

TimLued, 2021-03-23:
Hello,

does Carla 0.9.11 support the increase of car light brightness? or of the illuminated area?
The following picture shows the cyber truck model with light_state = vls.Position | vls.LowBeam | vls.HighBeam

[image]

Best regards

Madecu, 2021-03-23:
Hi @TimLued,

Unfortunately through API you can only turn the vehicle lights on and off. 
We are improving our vehicle gallery and we will implement the intensity and color customization in the future. We want them to work as streetlights and buildings do.

For now, you can edit the vehicle blueprint and change the intensity and attenuation radius for each vehicle.

[image]

TimLued, 2021-03-23:
Thank you @Madecu !

closed 2021-03-23 by TimLued