carla-simulator/carla #4010: opendrive could not be correctly parsed (WIN)

Leviathan321, 2021-03-23:
I have the problem that carla can not parse the .xodr file when executing:
`
py -3.7 config.py -x opendrive/TownBig.xodr`

I have seen there has been already some issues opened but only for linux.

I am using the latest night build windows version.

corkyw10, 2021-03-29:
Hi @Leviathan321 can you post the full output of the error from the console please? Can you confirm that the path is correct? Can you check that `CarlaUE4/Content/Carla/Maps/OpenDrive/OpenDriveMap.xodr` matches `TownBig.xodr`? If not there may be an issue with encoding which should be utf-8 

@Axel1092 could you follow up if needed?

Leviathan321, 2021-03-29:
Hi, my output is:

```
C:\Program Files\Carla\PythonAPI\util>py -3.7 config.py -x opendrive/TownBig.xodr
load opendrive map 'TownBig.xodr'.
opendrive could not be correctly parsed

```
There is no OpenDriveMap.xodr in the path, there are just TownXY.xodr files. I tried to run the files, but I get the same error.

corkyw10, 2021-03-30:
With another user we found that the issue was happening due to lack of write permission for the server on the directory  `CarlaUE4/Content/Carla/Maps/OpenDrive/` because the server needs to write a file there during parsing. Can you check if there are write permissions on that directory?

Leviathan321, 2021-03-31:
You were right, the program had no permissions. It would be good if this could be somehow logged in future.
Thank you Corinne

closed 2021-03-31 by Leviathan321